DroneBridge/DroneBridge #3: Install on RPI3 / Omnibus F4 FC

diyquad, 2017-11-29:
Hi,

I discovered your project few days ago, seams interesting!.
I tryed it, with your RPI Image, but it doesn't work for me yet.

Here is what i have, if you have an idea of what's wrong: 
- RPI3 with your Image on it
- Pi camera
- Omnibus F4v3 Flight controller  - with Inav last version  1.8.1 
- Wifi Key (classic key as i'm only testing for now)
FC is powered by Lipo. I use the USB to link it to RPI.
For now the RPI is powered with 5VDC 2A , and will be powered with BEC on the quadcopter.

- When i use your image, if no Pi camera connected, it launches, and i got a screen with horizontal and vertical "array", and some data. (like in an OSD) No image (since Pi camera not wired), and all data are 0.

- When i wired the Pi Camera, it stays on the ssh and display a lot of !!!! with the number of packet send. This number increase, as the  !!!!! increase also and take all screen.

Do you have an idea of what i did wrong? ;)

Thanks

seeul8er, 2017-11-29:
Hi! Thanks for trying out :)
First things first: 
- You don't need a GoPro. GoPros are not supported yet.
- DroneBridge is designed to deliver data over long distances using two Raspberry Pis (air and ground) with specific wifi adapters
- If a camera is connected the software automatically loads the "air" (rx) modules

As I see you do not need/want the long range link and therefore the provided software is not fit for your purpose.

However:
In what data of the FC are you interested in? (Telemetry? - Maybe I can be of help) 
As DroneBridge reads data from the FC you might be able to harvest some code from this project and modify it for your needs.
My best guess is you want something like: 
FC  --> Pi with cam --> App

diyquad, 2017-11-29:
Hi

Thanks for your quick answer ;)
Yeah you're right. For now, it's not really long range link i'm looking for, but as you say something like FC=>RPI +cam => App 
Since i'm using FPV for now, i don't really go high distance, but your project still interest me for my second part (will come in 2 month)

As you said, some of your code could help me.
Mainly get data from FC to RPI, so i can manipulate them in python.
Went to check your db_comm_rx.py and db_telemetry_rx.py but for now too complicated for me to extract what i need and make my little code. Is those files the right one for understanding how your communicate with FC?

Also, if i want to use your "complete" code, with 2 RPI (One PI2, One PI3), can i use another wifi key? (have some spare ;)

thanks again

seeul8er, 2017-11-29:
What do you mean with "wifi key"?

`db_comm_rx` does not communicate with the FC. It is just changing local settings.
`db_telemetry_rx` reads data (LTM telemetry or MAVLink) from the flight controller and sends it over the long range link. 
Take a look at the current development version. It is cleand up a little and supports MAVLink or any other kind of protocol coming from the FCs serial port:
[Development Repo](https://github.com/seeul8er/DroneBridge_Comm): The magic happens [In this File](https://github.com/seeul8er/DroneBridge_Comm/blob/master/db_telemetry_air.py): 

1. Open a socket to read incoming data from FC
```
import socket
import serial

def openFCTel_Socket():
    ser = serial.Serial('/dev/ttyAMA0', timeout=None)
    return ser
```
```
tel_sock = openFCTel_Socket()
time.sleep(0.3)
```

2. Now you can read the incoming data one byte at a time `byte_read = tel_sock()` 
3. Or a lot of bytes (e.g. 50 bytes) with just one call: `bytes_read = tel_sock(50)`

```

diyquad, 2017-11-30:
Hey

Thanks for your help.
I tryed what you said, i always get a 
TypeError: 'Serial' object is not callable
tryed different serial type (AMA0/ACM0/USB0/S0)

anyway, seams more a python problem and not your project ;)
thanks for help!

closed 2017-11-30 by diyquad