DroneBridge/DroneBridge #6: Runcam split v2 as main camera

Darozas, 2018-01-08:
Hi 

Thanks for your time to reply previously.

Can you make this to happen???

Can you make analogue video as the main source ??? 

> •make DroneBridge more indipendent of WifiBroadcast (video) > support of esp32 or esp8266 modules to allow integration in existing builds using analog video > no need for raspberry pi

Is that make sense ???

Regards

seeul8er, 2018-01-10:
Hi,
at the moment I do not have any plans on supporting analog video simply because I do not have access to such hardware. However by the end of this year I might will.
There would be two ways of integrating analog cameras:
1. Analog video grabber to AirPi and feed into digital downlink stream
2. Analog downlink using standard hardware and an analog video grabber connected to the phone via USB-OTG.

In the end it all comes down to analog-digital conversion latency. I think our best bet would be to do the conversion on the ground. That way we get the lowest latency possible, at the sacrifice of quality and range.
The thing with the ESP-Modules was to allow people to use DroneBridge (no video downlink!, just telemetry and uplink) without having to buy raspberry pis etc.
Analog video input options to the app would fit into this concept very well.

Darozas, 2018-01-10:
Ok thanks for your time and effort you put in this project keep it up

Regards

closed 2018-01-13 by seeul8er