DroneBridge/DroneBridge #7: Support 433MHz modules for Data Transfer

iRazoR, 2018-01-12:
Would it be possible to support 433MHz Connection Modules like HC-12 or 3DR Modules like this one:
https://www.banggood.com/3DR-Radio-Telemetry-433MHZ-Module-For-APM-APM2-Europe-p-85233.html?rmmds=search&cur_warehouse=CN

I think the potential Range is higher than Wifi.

geofrancis, 2018-01-12:
Yes but those can't do video so what's the point?

iRazoR, 2018-01-12:
I was just thinking it might be a good alternative for Telemetry only mode because of better Range than Wifi.

seeul8er, 2018-01-12:
I assume this is a request for the app? (Module - > USB - > Phone)
My answer is maybe in the future. The next two months I will be hold up by finals at my Uni. Very little time to code. Plus I do not have access to the hardware. 
The main goal right now is to make all modules as stable and easy to use as possible. I recently added uplink capabilitys to the control module. When the debugging is finished the projects main skeleton stands and I can focus more on the app. Then stuff like usb telemetry and analog video might have a chance of being added.

If you can not wait you could code yourself a serial to udp kind of bridge. The app accepts telemetry streams over UDP. That is the great thing with modularisation 😃

iRazoR, 2018-01-12:
Fair enough. Good luck at your finals and let us know how we can support you as you started a great project in my opinion.
The request was just started because Im investigating the possibilities at the moment and I didnt know that you cannot transfer Video over 433MHz Modules.

Thank you also for the hint with the bridge. I will definetly have a look on it. Shouldnt be impossible I guess.
That request can be closed

closed 2018-01-13 by seeul8er