DroneBridge/DroneBridge #9: Implement SBUS as a universal 8+ channel RC control protocol

careyer, 2018-01-16:
Hi there,

is it already possible to use MavLink for RC control? Since MavLink RC override is only limited to 8Ch it would be very useful to have SBUS implemented as a truly universal 8Ch+ RC signal. Thanks in advance!
I know of someone who has already made some progress in implementing SBUS.... however this solution still freezes after some time. Let me know if you are interested in exchanging some knowledge about this.

Cheers!
Thomas

seeul8er, 2018-01-16:
I was thinking about something like this. With iNAV I think standard RC still is the best option as we have no spare serial port on the Pi.

SBUS would be nice. Well supported with iNAV and MAVLink. Biggest problem with SBUS will be the baud rate of 100 000. Raspberry Pi does not support this one out of the box :(

How about SUMD? [Seems to be supported by Pixracer](https://pixhawk.org/modules/pixracer)
Will only give 8 channels with iNAV (not sure about MAVLink) but I do not see that as a big deal breaker for now. Just need to find some documentation for it.

MSP_RC is currently broken. Still don't know why it is not accepted. Worked just fine in the prev. release :(

careyer, 2018-01-17:
A friend of mine has implemented a proof of concept for SBUS on Raspberry Pi. He is not using the only serial port (that is reserved for MavLink Telemetry) but has implemented some kind of software serial port.. he has also struggled with the 100 000 baud thing but could solve it.
Moreover his solution consumes only very little CPU cyles since he his working with interrupts and DMAs.
Let me know if you are interested in taking in a look into the code. I will try to bring the two of you together then. Cheers!

seeul8er, 2018-01-18:
Thanks! If the code is stable I might wanna take a look at it. 
As for now I do not want to overcomplicate things. I guess I stick with SUMD. It just offers everything needed without using hacks etc. If someone wants to use it they will have to get a FTDI Adapter to get a second serial port if their FC does not support USB. I think that is as clean, easy, fast and stable as it can get.

careyer, 2018-01-18:
Unfortunately the code is not finished yet and will sometimes crash for unknown reasons. It still needs some work but we could already observe that it works quite nicely. Take Care!

seeul8er, 2018-06-10:
Hi, I'm back!
Just updated the code and we now have 12 channel SUMD (AirPI to FC) support. Set rc_proto=5 to use DroneBridge RC for the long-range link and enable SUMD output on the UAV on a defined UART/FTDI.

SUMD is universally supported with 12 channels by both iNAV & MAVLink based FC.
The code will be part of the next release.

As SBUS is not necessary anymore and will not be added by me in the near future I am closing this issue.

closed 2018-06-10 by seeul8er