DroneBridge/DroneBridge #10: Pi Zero Testing

cglusky, 2018-03-29:
Plan to do some wifibroadcast testing with a Pi Zero W with external RT5572 wifi dongle.

Would also like to test DroneBridge.  Any thing in particular you would like documented?  And/or anything to watch out for using a Pi Zero?

seeul8er, 2018-03-30:
Hi!
I have never tested a Pi Zero W but think it's cool you would like to try it out!
I have nothing particular in mind right now but keep me posted if something goes wrong :)

I hope to have DroneBridge v0.3 ready by next week with some MavLink telemetry support. Right now the ready to use images only do LTM telemetry and MSP.

cglusky, 2018-03-31:
OK.  I'll wait for v0.3.  Also waiting for some hardware anyway.  Will report back here when I give it a try.

cglusky, 2018-04-10:
Thanks!  No worries about timeline.  Still working on frame design.  This is for digital FPV on a 300 class frame.

https://github.com/goodrobots/DevFrame

https://photos.google.com/share/AF1QipPvirsbW9s801tFC5OBGKWGPwsN6hmORJvLOSQybhK2zXgq0uHHtkcFc3QRWDY3wg?key=SlV0NTJNVWdJQUlxb0ptLUtQVzBHTDUwR3RLT25B

cglusky, 2018-04-19:
Very impressed with video quality!  I see why the wifi broadcast community is so excited.

Setup:
TX:  Pi0 with OEM 5572 WiFi card; also connected to PixFalcon via ttyAMA0
RX: Pi3 with Alfa 051NH and AP Hotspot via internal wifi

Results:
- Solid video via HDMI to monitor on RX 
- NO Mavlink telemetry that I could see either via HUD in monitor or QGC
- Had video in QGC once but never got it back after a reboot of TX and RX systems  

Overall, I am a complete wifi broadcast newb.  I followed your instructions in the wiki for the ini settings.  Only thing I changed was channel to 5825 and reduced video to 30FPS and stream rate to 12mbit since I am running a Pi0.

Serial settings for control module were ttyAMA0 and 19200 baud to match FC side.  I tried both Mavlink v1 and v2 for protocol.  I also tried changing telemetry define in OSD config file with no change.  Really not sure what files need to be touched to tweak everything.

Also tried the android app on a google pixel.  Opens and then crashes when connected to RX hotspot.  Assuming I can get an IP assigned.

Hope that helps.

seeul8er, 2018-05-03:
Damn. What android version are you running? Did it crash every time right after startup?

I will fix the MAVLink thing in the next release. It probably is the same issue the guys from Wifibroadcast have with parsing the stream. Next release will have a plain pass through option. That will solve it.

OSD config just tells the OSD what data to exxpect and what stuff to display. Wifibroadcast config is for all the video and transmission stuff. DroneBridge config is for data links, RC etc.
I know it's a mess. Maybe I can solve it in the future

cglusky, 2018-05-03:
Pixel XL running 8.1. Yep. Open and then crash right after.

seeul8er, 2018-05-10:
I added a crash reporting system to the app. If you want I'll post a link here so you can install (& crash) that one. Maybe that gives me more insight on whats causing the problem :)

cglusky, 2018-05-11:
Happy to test it.

cglusky, 2018-05-11:
Also if you want to mark a 0.3.1 release or some such with your recent MAVLink pass-through changes i can test that as well if you like.

cglusky, 2018-05-12:
App is no longer crashing my Pixel.  Will need to hook everything back up and see about video feed etc.

seeul8er, 2018-05-16:
Made a new release. Tested it with iNAV. 
Set the `serial_int_cont` to the serial port connected to PixFalcon and just set the `serial_int_tel` to something that does not exist or disable it on the AirPi by setting `en_tel=N`.

in osdconf.txt set telemetry type to MAVLink

cglusky, 2018-05-18:
Great!.  I have to rebuild my 300 frame and will give it another try.  Might take me a week to get back to it but it is on the list.  Thanks!

BTW, It's not going to be Pixhawk/MAVLink so I won't be able to test that part in short term.  Going to stick to CleanFlight for my 300 for now.  That may change as ArduCopter 3.6 matures since it has more support/docs for F4 type flight controllers.

closed 2018-08-27 by seeul8er