DroneBridge/DroneBridge #12: Not an issue: ESP32 support is awesome! Any documentation on this?

careyer, 2018-06-27:
Hi,

just got notified about the new 0.4 release. From the release notes I derive that now the ESP32 plattform is also supported. Can you maybe update the Wiki and provide a bit of documentation of how to get this running on a ESP? Is this a full-blown solution, e.g. support for RC control and MavLink?

Thank you very much!
keep up the excellent work

seeul8er, 2018-06-27:
Hi,
at this moment you can connect to the ESP32 via WiFi with any kind of device. DroneBridge app is supported out of the box. I have also tested mwptools successfully with it. Other ground station software like QGroundcontrol or Mission Planner should work as well.
It supports MSP & LTM. MAVLink is also supported via a transparent pass-through option.

ESP32 will forward all data received via UDP on port 1607 directly to the UART. All data received via UART will be parsed (MSP/LTM) or directly forwarded (MAVLink/Transparent) to a connected WiFi device. That means that all functionality of the DroneBridge control module (except DroneBridge RC protocol & SUMD) is implemented. Video is not supported.

So yes you can use it to send MSP_RC or MAVLink_RC messages to the flight controller. However, DroneBridge RC is not supported right now. I do not see the point as almost every RC out there should have more range than ESP32s WiFi.
I'm also thinking about an ESP32 to ESP32 connection using WBCs technique or the ESPs custom ESP NOW or "LR" option. But it seems that the range is not that much better so it's not getting the highest priority right now.

[The code/firmware with some additional info can be found here](https://github.com/seeul8er/DroneBridge/tree/master/db_esp32). I will add some info to the wiki during the next days.

Nice to see that there is some interest :)

careyer, 2018-06-30:
Thank you very much for your timely answer. Too bad that video is not (yet) supported. Maybe you could team up with bitluni? He has excellent skill when it comes to using ESP32 in combination with cameras and video. You might want to check out his homepage and youtube channel. (http://bitluni.net/) 
As far as I know he is german. So cooperation which makes cooperation easier in a way ;-)

Have a nice weekend!
Cheers!

seeul8er, 2018-06-30:
DroneBridge for ESP32 is intended to be used for ultra cheap telemetry/bi-directional links. The idea was that it can be used in combination with analog video transmitters that people might have lying around. It is an easy and cheap way to get started.

Due to the low data rate of the SPI/I²C/UART ports it is not possible to use the ESP32 instead of a USB WiFi adapter or with a camera module directly connected to it.
The projects of bitluni suffer from the same problem. He can only get some frames with relatively high latency through the air. bitluni seems to know his stuff, but I do not think he is aiming as high as DroneBridge when it comes to fps & latency

closed 2018-07-15 by seeul8er