DroneBridge/DroneBridge #13: How to build this from scratch

ewized, 2018-06-28:
How do you build this project from scratch? Trying to add custom code to the profile and add extra programs to the image but currently ran out of space and need to reimage the SD cards. Wounding how you would build this program.

seeul8er, 2018-06-29:
You will need set up a new raspbian installation with a patched kernel. You can find the patches and some instructions in the wiki of EZ-WifiBroadcast.

I would not go that path. If you need all the space of your SD-Card [just expand the filesystem](https://www.raspberrypi.org/documentation/configuration/raspi-config.md) using `rasp-config`. That should do it.

ewized, 2018-06-29:
I tried expanding the file system though it said it was expanded the OS is still using 2GB (1.7GB).

ewized, 2018-06-29:
There was an error and ran this command after expanding to fix the problem. Thanks.

```
sudo resize2fs /dev/mmcblk0p2
```

closed 2018-06-29 by ewized