DroneBridge/DroneBridge #17: EZ-Wifibroadcast: "The RC implementation is missing important security features"

rodizio1, 2018-07-31:
Which important security features would that be?

careyer, 2018-07-31:
Maybe seeul8er refers to this:

> The RC implementation is missing a check wether the data arrived correctly on the drone. Only eight channels are supported. (source: http://wolfgangchristl.de/dronebridge/)

Was "The RC implementation is missing important security features" before (source google cache)

seeul8er, 2018-07-31:
Changed it like 3min ago :) Wanted to make things more clearly. 
I removed it because the dramatic look of "security" when all that's missing is a CRC check of the data. At least I couldn't find one. 
Don't get me wrong. It's not like the drone is going to fall out of the sky because of that, but it still is a little weakness of the protocol.

I really appreciate the newer OSD and Kernel patches. It's not really a different direction we are going, just a different way of doing things.

rodizio1, 2018-07-31:
> "the RC implementation is missing a check wether the data arrived correctly on the drone. 

There is no "data arrived check", because:
a) It's only supported by the MSP protocol (not because it makes sense though, simply because the MSP protocol simply acknowledges every message it gets). The MultiWii guys realized years ago that it's pointless (see below) to generate ACKs for R/C messages and removed it, see here: http://www.multiwii.com/forum/viewtopic.php?f=8&t=6231)

The inav guys also eventually realized this, see here:  https://github.com/iNavFlight/inav/pull/2762

b) It's pointless, just wastes CPU cycles and just adds a lot of complexity, and room for error and bugs:
Ask yourself: How do you determine that a message was lost? You wait for the ACK, right? For how long do you wait? Considering, that EZ-Wifibroadcast R/C allows for about 150Hz or so, you only have 7.5ms after the next message arrives. Since Linux (and especially the Pi with it's underlying firmware) is not a realtime system, it can well be, that your code in userspace (i.e. after the scheduling inside the flightcontrol, after the buffers in the flightcontrol, after the transmission over the serial wire, after the serialport FIFO buffer on the Raspberry, after the Raspberry firmware and it's non-realtime scheduling, after the Linux kernel and it's non-realtime scheduling, after your code reaches the part where it checks for the ACK) simply gets the ACK too late, and you wrongly detect that the message did not arive.

The next question would be: What do you do with that knowledge? Re-send the message? Pointless, as it will be outdated soon anyway, the next message comes only a few milliseconds later. There's no point in sending stick-positions "from the past" to the flightcontrol. Warn the user? Well, could be an option, but if R/C is lost, the user will notice very quickly anyway. The only error that could be detected that way is intermittent message loss on the serial line, but only when using MSP and a flightcontrol with a MSP implementation that actually sends MSP ACKs.

 

> when all that's missing is a CRC check of the data. At least I couldn't find one.

You obviously overlooked the 802.11 FCS. This is basic knowledge.

The only places where a CRC check makes sense is on a transmission path that can lead to corrupted data. That would be: 
- The serial line, which is protected by the R/C protocol CRC implementation already
- The wifi transmission, which is protected by the 802.11 FCS (32bit CRC) generated by the wifi card already.

So the extra one-byte CRC inside your protocol is completely pointless and actually makes things worse because it increases the frame length and thus the probability of packetloss. It's also wasting CPU cycles for the computation.

rodizio1, 2018-08-01:
> The FCS check (fcsfail) should be disabled since we have forward error correction.

This has really nothing to do with the forward error correction in Wifibroadcast, the FEC doesn't really "protect" against frames with wrong FCS. Actually, Befinitivs idea was to let packets with bitflips go through (in cases when all FEC packets have been "used-up" already).

> You could argue now that disabling it will only make sense on the VideoRX since all the data sent to the UAV is generally not protected by FEC in DroneBridge (Not so sure about your WBC stuff). And you would be right. I did not check what the current setting is and because of that I may remove that statement from the website and apologize for the formulation. It still is a weakness of the implementation but not if the system is configured the default way.

Befinitivs idea behind letting frames with failed FCS check through was to get as much data as possible through without losing too much. I.e. "don't drop a 8000 bit frame just because there is only 1 bit flipped". This is, in theory, indeed a good idea.

However, the problem with disabled FCS check in practice was (I have done extensive testing ...):

a) Sometimes, something inside the kernel or wifi card (never analyzed where exactly it happens) "hangs" when too many packets with bitflips are received, leading to _no data reception anymore at all until reboot_

b) Sometimes, something inside the h264 decoding pipeline hangs, leading to a _frozen video until hello_video.bin is restarted_

So I argue that it is **very dangerous** to disable the FCS check as it may lead to _complete communication loss or video freeze_.

This is why in my images, frames with wrong FCS get dropped (and also the code inside the rx program regarding this is commented out/disabled to save some CPU cycles, as there is no need to check for that when frames with failed FCS already get dropped before they reach the rx program).

> It still is a weakness of the implementation but not if the system is configured the default way.

No, it's not a weakness of the implementation, let me repeat, it was a deliberate choice, since an additional CRC done in software:

a) Adds another byte to the payload, increasing packetloss both on the R/C and videolink, because the probability of a collision goes up the larger the frames get (I have done extensive testing on that ...)
b) The frames are already perfectly protected by the 4-byte 802.11 CRC, and the Pi0 is not that fast, saving CPU cycles is always a good idea, even if it may not be that much