DroneBridge/DroneBridge #21: Release 0.5 - Misssing telemetry between air and ground

NanoCarbonZA, 2018-08-05:
Hi. Ive installed the latest version and there seems to be an issue withe the teletry between the air and ground unit. It prob something im missing but canr get it to work. Ive also tries the app via wifi hotspot and usb. 
Via Usb i cant talk to the air or the ground unit
Via hotspot it seems to fail to talk to the air unit. 

I do receive the video on hdmi usb and wifi hotspot. Can you please point me in the right direction. 

Just as a fyi... If the PI's is under voltage the bitrate drop and the quantity of the via goes with it. Due to the telemetry i didnt see the air unit was underpowered.

Telemetry is working on ez-wifibroadcast.

seeul8er, 2018-08-05:
Have you plugged in any USB devices like an RC, USB stick or keyboard (except wifi adapters)?
Can you see the telemetry in the OSD?
Are you using MAVLink or MSP/LTM? (Did you configure DroneBridge it for MAVLink?)
Did you change any settings in the wifibroadcast config except video and wifi settings? (Like telemetry ports etc.)

Thanks to some people testing the app a am able to fix some bugs. Maybe that is one of the issues.
Lot's of questions :smile:  Hope we can get behind it.

NanoCarbonZA, 2018-08-06:
I have not plugged in any other devices and started with a standard installation with now changes.
The setup is
AirPi ->usb 2.4 ghz wifi dongle -> usb 2.4 GHz wifi dongle -> GroundPi -> HDMI or Android App.

On the GroundPi's OSD, the CPU and Temperature is populated for the GroundPi but not for the AirPi.
Also the under power message for the AirPi isn't displaying on the OSD screen.

With both not appearing, it makes me thing that the telemetry data is working. Still need to connect an actual flight controller (INav and Pixhawk)

After that i want to test the FrSky  RC.

BTW... Will to assist if needed. I busy familiarizing myself with the code.

seeul8er, 2018-08-06:
The absence of the AirPi CPU temperature and usage is not a bug. I believe I removed the call to the wbc script that does that, so it's not starting. As the AirPi should always be cooled pretty well due to the natural airflow around the UAV, bringing back the feature has no big priority to me right now.
It has nothing to do with the telemetry module (or any other module). They should work just fine.
I am thinking about bringing the information back via a custom DroneBridge status message.

However, you should be able to receive the settings via the app. I fixed some bugs and put out the link to the new version. Maybe that fixes some issues.

Thanks for testing it! The learning curve still is very steep. I will try to add more documentation this week. Especially about how to set it up, wire it up and some stuff about the modules for developers.

seeul8er, 2018-08-07:
Jep. Just flash the image. You can connect via USB to the ground station. I recommend that you plug in the cable after the ground Pi has booted.

> The dronebridge app is only capable of changing settings on the ground side i.e. Telemetry, FEC settings Etc...

Where did you read that? (Need to remove it) It's not true. It can change the settings on both Pis. It just _requests_ the WBC settings from the ground Pi only.

careyer, 2018-08-07:
Hi Wolfgang

> The absence of the AirPi CPU temperature and usage is not a bug. I believe I removed the call to the WBC script that does that, so it's not starting. As the AirPi should always be cooled pretty well [...]

It would be nice if you could add this feature again. This was a very useful feature especially when using a Pi0 as the AirUnit. Once you throw a bit more at at Pi0 (like a 2nd camera) its' computer power capacity is reaching the limit quite easily. Once this happens it starts thermal throttling. Also when using it on a rather stationary multicopter (hovering) cooling is not as good as in fixed wing aircraft. If it ain't too much hassle it would be nice if you could bring that feature back ;-) Thanks!

ewized, 2018-08-09:
Same I would like to see the OSD info of the AirPi as well its very useful in testing.

htcohio, 2018-08-09:
Hey guys,

I finally got dronebridge working!!!!

 I'm able to change all the video and frequency settings from the Android app and my setup is with pixhawk and lrs dragonlink for now but I can confirm seeing the channel values on the Android up so RC control should not be a problem.

@Seeul8er you really did a spectacular job on this and must have put in a ton of work. I can't wait to field test it.

I do have a few questions or things to bring to your attention.

1. I have a Samsung Galaxy S8 that has virtual buttons that pop up from the bottom which you can see in the video.  I either have to close the app or wait what seems to be an eternity for it to go away.  Anyway you could move the gear icon to the top somewhere? I believe the icons for the map function are also covered, perhaps those can be shifted to the left a little as well?

2. I tried using my video switch, I think I broke my other one and I just received a new one today and that is not working at all.  Since user1321 is the one who got that up and running for me you may want to coordinate with him because, all of the lights come on with the cameras which is a good sign! I'll also add that I've noticed a problem arises if you try switching to a camera Port where no camera exists or pwm levels were not set correctly.  You basically have to restart the air unit.

3. Could you clarify the recommended rebooting or reset procedure after settings were changed in the Android app? 

4.  Would it be possible to have a "reboot" air and/or ground function in the app?  If rebooting the ground is necessary the only drawback is eating to go through enabling USB tethering and all of that which is no big deal compared to fumbling around with SD cards but it would be a nice enhancement.

I sort of threw together a video of my setup tonight.  I apologize for the quality I was just trying to get the information across.  **FYI** towards the end of the video I mentioned that the video resolution you are seeing appeared to be a bug, this was not the case that was just me being an idiot because I didn't set the correct resolution for the settings LOL

I also attached a link to my Dropbox that contains all of the text files as I had them.  I've also modified the OSD by Shifting the ladders outwards and changing the font a little bit.

**Notes**
>>The biggest problem I missed while setting this up for the first time was changing from "LTM" to "Mavlink" in the OSD text file. (I misssed this because the original wfb image was set to mavlink).

>>The other problem I solved was that the Android app had Port "1604" as default under the app settings tab... After I changed that to 1605 the app started to communicate with the ground station.

Please let me know if you have any questions and I'm also going to copy this post on GitHub for the project.

https://youtu.be/2Dccs2EPnkg

https://www.dropbox.com/sh/j1h3aboel9cqltd/AAB4jvhMCtKWMS4Ia6DDXzY7a?dl=0

Cheers!

My modified OSD settings (included in Dropbox link)
[image]

seeul8er, 2018-08-09:
Cool! Very nice :smiley: 
The port should be set to 1605 by default, but good thing you mentioned it.
For all the others: Make sure you read the getting started guide.

> I have a Samsung Galaxy S8 that has virtual buttons that pop up from the bottom which you can see in the video. I either have to close the app or wait what seems to be an eternity for it to go away. [...]

The icon stays where it is :smile:  I will try to make the navigation bar less annoying.

> I tried using my video switch, I think I broke my other one and I just received a new one today and that is not working at all. Since user1321 is the one who got that up and running for me you may [...]

I will have a look at his code. Hoped to get it right the first time since the docs were so clear.

> Could you clarify the recommended rebooting or reset procedure after settings were changed in the Android app?

Wait for the positive feedback inside the app, wait 1-2 seconds and then unplug and replug both Pis. It could happen that the power supply has a big capacitor that needs some time to discharge when disconnected from the main power source. During that time the Pi might still be powered although you pulled the plug. Wait a few seconds (till the Pis LEDs went out) and then repower.

> Would it be possible to have a "reboot" air and/or ground function in the app? If rebooting the ground is necessary the only drawback is eating to go through enabling USB tethering and all of that which is no big deal compared to fumbling around with SD cards but it would be a nice enhancement.

Thought about that couple of times but I think it is too dangerous. Anyone in the area could send out rough commands. Without encryption (and even with), there is no way of telling whether the command was issued by the pilot or not. Right now the community is very small and I never really cared about security too much, but adding a feature that will power off the the Pis in any situation is a little too much.
Additionally, the Pis sometimes do not reboot properly. My ground station eg. never detects the wifi cards after reboot. That may be an issue with the power supply or something else but imagine that happening in the air.
I am however thinking about changing the scripts so that some settings take effect without requiring a reboot.

> Same I would like to see the OSD info of the AirPi as well it's very useful in testing.

Ok ok. Will see what I can do ;) Shouldn't be too much work.

seeul8er, 2018-08-11:
Just release v1.2.4 - Check out if the new behaviour of the navigation bar works for you. It should hide now if the app is in focus.

I am not sure if it is a good idea to let the user set the tx power via the app. In general, you do not need to change it that often. You just try around till you get the setting that works best for your card. After that, you will just stick with it. I consider the tx power an "expert setting". You can really make things worse by overdoing it. Newcomers will just see the parameter "tx-power" and then be like "lets ramp that thing up" because more is better right ;)

Besides: According to Rodizio the Atheros cards are already operating at max. Changing the tx power is only advisable on specific Ralink cards.
I think that adding that information to the "getting started" section is the better idea. Together with some technical and legal info about tx-power and EIRP.

If I got some time left after adding more critical features I could think about an option inside the app that does this. But not at the moment.

htcohio, 2018-08-11:
Perfect timing!

htcohio, 2018-08-11:
New app looks really good but trying to look at the source code to get the
camera switcher working.  My board could be bad from the factory but we may
try to use the one that was in a crash trying to figure things out will
update soon.

htcohio, 2018-08-12:
Hey @Seeul8er,

User1321 Forks remotely with me today to try and get the video switcher up and running.  I believe he wrote some additional script but you'll have to get with him on that.

No matter what settings I try to change we keep getting an error that says "control module and Telemetry module are assigned to the same serial port". 

I'm going to post a picture of the screenshot on rcgroups forum. 

I've tried to disable control on both ground and are sides and a triple checked all the parameters which are copied from the previous working version.

Is there anything that could be causing this problem? 

Also, as for TX power I see your point.  I might Ubiquiti Wi-Fi stations I actually drop it down from the default setting of 30dbm to 28 for the best performance but you're right it's a advanced function.

Thanks

htcohio, 2018-08-12:
Please disregard this for now.  I think I have the switch mostly working however, I have 3 RX cards and I only seem to receive video when I set ground to "auto".  If I try setting control and Comm two specific Mac address I don't get any video.  Am I supposed to set Mac on video and telemetry with them being disabled modules?

seeul8er, 2018-08-12:
>  If I try setting control and Comm two specific Mac address I don't get any video. Am I supposed to set Mac on video and telemetry with them being disabled modules?
No. Not necessary.

> I only seem to receive video when I set ground to "auto"
Strange. Video is supposed to be completely independent of all the other stuff. Config-wise and code wise. I'll see if I can reproduce the problem

NanoCarbonZA, 2018-08-12:
Hi

I'm still have issues with the OSD and displaying info from the INav flight controller.
The only way it works for me is when i set 
wifibroadcast-1.txt
    TELEMETRY_TRANSMISSION=wbc
DronebrideAir.ini
    tel_proto=mavlink
    baud_tel=57600
osdconfig.txt
    #define MAVLINK

INAV
  feature TELEMETRY
  port = mavlink (57600)

Something I'm missing here?

Also, Ive read this under the DronebrideAir.ini file
# serial interface used by telemetry module - telemetry module is DEAKTIVATED! THESE SETTINGS HAVE NO EFFECT

seeul8er, 2018-08-12:
So you want to use MAVLink telemetry?
Set all settings to default (you can get the default files from this repo) on both Pis. Also the WBC config file.
osdconfig.txt
```INI
#define MAVLINK
```

ground.ini
```ini
- no changes to the default settings
```

air.ini
```INI
en_tel=N
en_comm=Y
en_control=Y
serial_int_tel=/dev/serial1
serial_int_cont=/dev/serial0
baud_control=57600
serial_prot=5
serial_int_sumd=/dev/ttyUSB0
```

WBC-config
```INI
- no changes to the default settings
```

That way we use the dedicated UART of the raspberry pi. Connect that one to your flight controller.
`serial_int_tel=/dev/serial1`
`serial_int_cont=/dev/serial0`
is necessary because of a bug that will be fixed in the next release. The startup script does not allow it to have the same serial port set for both modules, even if one of them is deactivated.

All the modules work perfectly well. You must have a configuration error.

htcohio, 2018-08-13:
@seeul8er

I applied the changes to the code that you mentioned on rcgroups for the Telemetry changes eliminating spaces between the "=" for the two settings.

It looks like the OSD is back, power cycled everything several times to make sure it was consistent.

Strangely, the 5. 8 hotspot does not allow any Telemetry or control through the app.

I'm getting a IP Checker when I type alt+f4 on the ground side.

Any ideas?

patemil, 2018-09-10:
Hi,
Does it have to be Atheros cards to get rc-Ctrl working?

Have tried msp and mavlink, running iNav 2.0 and wifi dongles with Ralink chipset but I don’t seem to get anything through.
Video is working great in all conditions

careyer, 2018-09-10:
Yes! Needs to ne Atheros afaik.

closed 2020-09-16 by seeul8er