DroneBridge/DroneBridge #25: Feature request: Add support for MavLink v2 protocol version / remove 8Ch limit

careyer, 2018-08-10:
Hi Wolfang,

from what I understand from all the discussions about EZ-Wifibroadcast / Dronebridge is that the Mavlink implementation used right now is somewhat problematic. It also brings some nasty limitations with it e.g. limitation to 8Ch. From my research on the web I have learned that Mavlink protocol verison v1 is meanwhile considered a legacy standard and was succeeded by the Mavlink protocol v2, i.e: no one is maintaining v1 anymore and the current problems and shortcomings are likely to stay for ever or even get worse as more and more flight controllers use the recent v2 standard.

So I am wondering: Maybe it is a good idea to implement MavLink v2 support?

Lets look at an example: 
The limitation to only 8Ch was a limitation of the MavLink 1.0 protocol - especially a limitation of the `RC_CHANNELS_OVERRIDE` messages which are used for controlling the aircraft via EZ-WBC.

However this bottleneck seems to have finally been removed in Mavlink v2. :D
MavLink 2.x specifies so called extension fields which are added on top of the v1 protocol version. (They marked in blue).

[image]
[image]

So if we could go for the v2 Standard (maybe it also brings less problems than the broken v1?) we could luckly use 18CH - all proportional! :)

MavLink v1 is also officially considered to be a legacy standard:
[image]

This info can be found here: https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE
I am aware that adding and testing Mavlink v2 support is for sure a awefull lot of work but it might be worth it. I'd be more than happy to help testing and contributing in any way possible - sadly I am not a coder. :-/

Cheers!

seeul8er, 2018-08-10:
Hi,
MAVLink v2 support is part of DroneBridge since day one. I never used the v1 libraries.
With the control module set to transparent pass through (option 5) the protocol does not matter. Any message sent will get forwarded to the FC and vice versa. 

That must be a fairly new message (I think it did not exist some months ago). I will add support for it in the control module.

In fact there will be two ways of using it (like with MSP):
* 12ch support using to DroneBridge RC packet (MAVLinnk message generated on the UAV side)
* 14ch support when the message is generated on the ground

I know the MAVLink message supports up to 18 channels but I doubt that most people need more than 14 channels. More channels create more collisions and most of the stuff they want to achieve is possible using the communication module and single messages.

careyer, 2018-08-10:
Hi Wolfgang!

Wow! Did not know that Dronebridge already uses the MavLink v2 Libraries. That is terrific news! However I am happy that there was at least something new to my message after all (the new messages for more channels). :-) 

This will solve a lot of problems and inconvenience. And yes 12Ch (Dronebridge RC-packets) and 14Ch (transparent pass-through) is way enough.
Thanks for implementing the new messages!

Cheers!
Thomas

P.S: I am wondering if it might be possible to combine this with #18 ? Especially smaller Flight Controls (e.g.: Pixracer) have only very limited AUX outputs meaning that you cannot pass-through/route the additional channels out of your FC for example to drive some physical actuators.

Maybe with the Dronebridge RC-packet approach it is possible to declare somewhere (ini-File or App-Settings) that either all channels go to the FC or some upper channels (e.g. Ch9-Ch12) are split off and drive GPIOs instead? Just an idea.

careyer, 2018-08-24:
@seeul8er : If you need help in testing or anything please reach out to me! Cheers!

careyer, 2018-09-08:
Dear Wolfgang, I skimmed through your commits and as far from what I can tell this should be supported now, is that correct? Is there any way to test this already? Thank you very much and have a great weekend.
CheerS!

seeul8er, 2018-09-08:
[Posted a test image here](https://github.com/seeul8er/DroneBridge/issues/27#issuecomment-419596167)
My flight controller died so everything besides the video is untested for now.

On the Ground side you need to set:
`rc_proto=5`

On the Air side you need to set:
`serial_prot=4`

Everything is the same as usual. It might only work with a flight controller that supports that (newer) feature. The FC firmware needs to be compiled with the newer kind of MAVLink libs.
Happy testing :)

careyer, 2018-09-08:
AWESOME! Thank you very much... will download the image an start testing right away!

closed 2019-05-29 by seeul8er