DroneBridge/DroneBridge #26: Android app

user1321, 2018-08-10:
Good Day!
First of all, thank you for a great project!
I wish to do a minor changes to it, but I can`t find source codes of Android application, will you be so kind to share it?
Maybe e-mail or git?
Thank you in advice.

seeul8er, 2018-08-10:
As for the moment, my plan is to keep the app closed source until I no longer actively develop the project.
Right now there are so many changes in every version that having multiple version of it out there will just cause a lot of confusion and will result in incompatibilities in the short term.
The app as of its current state will always be free.
So I am afraid I can not help you with that one. In the future, I may hand out the code to some individuals that want to modify it for their purposes but not at the moment.

All the protocols etc. are open so if is necessary people can build their own little mobile apps that work with DroneBridge and for them.

closed 2018-08-10 by seeul8er

cgrrty, 2021-04-16:
Hello. 
I think there are still some people who are interested. Is it time for open source?