DroneBridge/DroneBridge #27: Feature request: Add support for recent wifi chipsets

Yes21, 2018-08-12:
The list of wifi cards supported by ez-wbc is pretty old, and didn't evolve since 2015 !!
Many of the supported cards are old, or not manufactured anymore (especially dual bands cards).
There are now very promising dual band long range cards from Alfa Networks :
- AWUS036ACH (RTL8812au)
- AWUS1900 (RTL8814au)

The architecture of ez-wbc is so old and difficult, that nobody could add support for now chipsets ...

(same issue [here](https://github.com/bortek/EZ-WifiBroadcast/issues/105))

seeul8er, 2018-08-12:
That is a very important point. The project stands and falls with support for Wifi Adapters. Unfortunately, I do not own one of these cards. I might get into it in September but I would be very happy if someone gets a patch working.

[Are you aware of this?](https://github.com/astsam/rtl8812au)

Yes21, 2018-08-12:
The best drivers are perhaps here : [https://github.com/aircrack-ng/rtl8812au](https://github.com/aircrack-ng/rtl8812au)

Yes21, 2018-08-13:
If you need help to buy these cards, I can help.

roque-canales, 2018-08-15:
I can also help you with Yes21

Yes21, 2018-08-15:
Do you know this project : [wifibroadcast-image-builder](https://github.com/RespawnDespair/wifibroadcast-image-builder) ?
It can create a wbc image from scratch using qemu, with the latest kernel and patchs.
The image is not totally functional now, but I will try to use it for testing aircrack-ng/rtl8812au drivers

htcohio, 2018-08-15:
Id like to bring up one more item related to a no issue with running four wifi cards  (3 Atheros for RX + 1 non atheros for the 5. 8 hotspot). 

Three-way diversity on the ground with atheros cards work perfectly fine by themselves however, if I connect fourth card for the hotspot and try to boot it freezes up after it loads the first or second card.

are there any possible Immediate Solutions to this? is there any way to connect 2 RX pis to eliminate this problem?  this would also provide a few extra USB ports for other things such as keyboards and USB app functionality

Yes21, 2018-08-15:
@htcohio 

I really don't understand why you've posted those comment here ...
It seams it has no relation with support for new chipsets !

Perhaps could you create a new issue [here](https://github.com/bortek/EZ-WifiBroadcast/issues)

htcohio, 2018-08-15:
Hey I'm sorry, I will open a new issue.  I've mostly been using an app called Fasthub and haven't been able to find out how to post a new issue until just now.

I will repost there.

As for the wifi adapters, I agree.  Do you know if there's anything in the work or that has any potential to have the frequency capabilities of the atheros AR9271 chipset cards?  I'm just wondering if I've missed something (and I haven't looked at the links above yet I will do that right now).

htcohio, 2018-08-23:
Hey guys,

I came across this info on wikidevi today about an atheros dual band AC wifi card.

Looks pretty promising, it is not very powerful but you can always amplify it.

https://wikidevi.com/wiki/SparkLAN_WUBQ-159ACN

802.11ac/a/b/g/n USB Module, Qualcomm Atheros QCA9377-7, 1T1R
- Chipset: Qualcomm Atheros QCA9377-7

Yes21, 2018-08-23:
@htcohio 

This chipset is not supported by ez-wbc.
While it is "not very powerful", I don't see the interest for getting support for it ...

Support for chipsets 8812au or 8814au, and powerful Alpha cards, would be more useful for the future (and the present) of ez-wbc and DroneBridge.

ps : that's just my opinion.

roque-canales, 2018-08-24:
+1 Yes21

htcohio, 2018-08-24:
@yes21,

I read in a couple places that the new atheros chipset is capable of monitor mode and ath10k driver.  There are a few variants of that chipset that are listed as new on the page. I figured it was worth mentioning

Yes21, 2018-08-24:
**@htcohio**

I'm sorry for being focused on realtek chipset and alpha adapters ...

There's certainly many other recent chipsets that should be supported.
You are right to mention it.

seeul8er, 2018-08-27:
@Yes21 I am a total noob when it comes to kernels and drivers so this might take some time. I will see if I can integrate the v5.2.20 branch of aircracks rtl8812au driver. [Rodizio reported that the prev version has issues with packet loss.](https://github.com/aircrack-ng/rtl8812au/issues/116#issue-312865808)

I might also upgrade the image to Kernel v4.14

Right now I do not know if I want to spend another 45€ on hardware. My source would be amazon. You got something cheaper in mind?

Yes21, 2018-08-27:
**@seeul8er** 
I'm a total noob too ! But I think that this project ([wifibroadcast-image-builder](https://github.com/RespawnDespair/wifibroadcast-image-builder)) could really help us : it can create a working image of ez-wbc 1.6rc6 with the latest kernel. It should not be very difficult to add a few lines to the building script to install the aircrack-ng/rtl8812au driver (you should also consider the v5.3.4 which support rtl8814 too).
I've already test his script, and I could generate an image in 26 min (I've a big pc)
I will try to ad the installation of the driver to his script, but I'm not sure to have success ...
If that way of processing makes sens for you, the guy "RespawnDespair" will perhaps be happy to help us ?

If you want a 8812au card only for testing, have a look at the alfa awus036ac which as the same firmware, but is "only" 29€. I will send you a MP on RCgroups on this subject.

seeul8er, 2018-08-28:
Damn, what release branch did you use? 5.2.20?
Is it still the packet loss?
Thanks for testing :)

Yes21, 2018-08-28:
branch 5.3.4.
I can't see loss packets, but my test is partial : awus036ach on a Pi3 as rx, csl 300 on a Pi2 as tx, freq 5200 mhz

roque-canales, 2018-08-29:
Yes21, could you share your image for I test, I have two setups ready with 8812AU on rx and tx side.

Yes21, 2018-08-29:
Sorry, but I can't share it because I have a slow internet connection, and it would be much too long for me to upload a 3,7 Gb file.
It's really very easy, even for a noob like me, to build the image following the instructions of this link ([wifibroadcast-image-builder](https://github.com/RespawnDespair/wifibroadcast-image-builder)) . If you want to try it, I can send you the 2 scripts I've modified to install the aircrack-ng/rtl8812au drivers.

zipray, 2018-08-31:
http://forum.rcdesign.ru/f90/thread515083-9.html#post7446277
http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.zip

Yes21, 2018-08-31:
**@zipray**
I thank you very much for sharing this image with us :)
I've just tried it, and I'm sad to write that it didn't saw any improvements in rtl8812/8814 chipsets.
- on the rx side : tha awus036ac and awus036ach are working, like with regular 1.6rc6 Rodizio's image. The awus1900 is not recognized.
- on the tx side : awus036ac, awus036ach and awus1900 aren't recognized.

Could other ones try this image, and confirm my results ?

Ps : I've also took a look at your corrections (diff file), and were surprised that you didn't need to make more modifications to ez-wbc code.

seeul8er, 2018-08-31:
Do you use the Pi Zero on the Air side?

Yes21, 2018-08-31:
If the question is for me, the answer is : no, I'm using a Pi2b.

zipray, 2018-08-31:
https://youtu.be/2sFczUCAKgQ

Yes21, 2018-09-01:
Here are the original links posted by zipray :

[http://forum.rcdesign.ru/f90/thread515083-9.html#post7446277](http://forum.rcdesign.ru/f90/thread515083-9.html#post7446277)
[http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.zip](http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.zip)

zipray, 2018-09-02:
@roque-canales [https://item.taobao.com/item.htm?spm=a230r.1.14.1.56fe61f37YzuL7&id=560527329875&ns=1&abbucket=1#detail](url)I don't know if you can buy it there, and whether it is in line with international shopping.

roque-canales, 2018-09-02:
Unfortunately,not.
Can I send you money for you order it and send it for me here in France?
my email kuspower@live.fr, I can pay through paypal.
Or onother solution, it that you give me the email of this seller for I attempt to contact him.

I hope you can help me for that ;)

seeul8er, 2018-09-02:
@zipray  What modifications did you make to the WBC image? What aircrack-ng/rtl8812au branch did you use or is it just the plain WBC 1.6 image with the built in drivers?

Yes21, 2018-09-04:
**@seeul8er**
Here is the link with the modifications : [http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.diff](http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.diff).

I'm pretty surprised that the modifications were made only in ".profile" and "tx_telemetry.c" ...

Yes21, 2018-09-06:
Today I've made news tests with this image : [http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.zip](http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.zip)

I have now very good results that I want to share :

- awus036ac, awus036ach and awus1900 are working on mi Tx Pi2,

- awus1900 is working on my Rx pi3,

- my csl300 (rt5572) doesn't work anymore with this image.

I've only tested video transmission (no telemetry, no RC) at 5180 Mhz, and didn't test the range.

Yes21, 2018-09-07:
Today I've made news tests with this image : [http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.zip](http://my.korchagins.ru/WifiBroadcast/EZ-Wifibroadcast-1.6RC6+8812au.zip)

I have now very good results that I want to share :

- awus036ac, awus036ach and awus1900 are working on my airPis : Pi2 or Pi0,

- awus1900 is working on my groundPi : pi3,

- my csl300 (rt5572) doesn't work anymore with this image.

I've only tested video transmission (no telemetry, no RC) at 5180 Mhz, and didn't test the range.

seeul8er, 2018-09-08:
@Yes21 
Yeah they probably installed it using dpkg. I am not sure but it could be that the wrong (unpatched) sources are used when it is installed via dpkg.
[I set up a separate repo for the DroneBridge Raspberry Pi Kernel so that everyone is on the same page and others have the chance to follow the steps.](https://github.com/DroneBridge/RPiKernel)
It contains the patches, config files for Pi2/3+ and Pi0 and under releases you can find the patched source code so people can directly modify and recompile the kernel.

I setup an experimental image that should support the Pi3+ and uses the current aircrack-ng/rtl8812au drivers.
You can download it here: [link to Beta v0.6](https://forstudents-my.sharepoint.com/:u:/g/personal/ga25puh_forstudents_onmicrosoft_com/EWi7-HH8Y1pOjLUn_ceffugBhC0Dsemiy5vxi_d9oduu-w?e=yJOoHS)
Let's see if that thing works. I only tested video using Atheros cards so far.

roque-canales, 2018-09-08:
@zipray please help us to find contact with the hdmi to csi2 seller, through taobao is not possible to order outside china….

Yes21, 2018-09-08:
I've tested your experimental image with all my Pis and all my wifi boards.
The image is correctly booting, but none of my boards are detected or set up, even the rt5572 csl300.

I want now to try DroneBridge with the "russian image".
Could you explain how to install DroneBridge on ez-wbc ?
It would be helpful if everybody could test DroneBridge on a modified ez-wbc image.

Yes21, 2018-09-08:
**@seeul8er**
I've tried your experimental image today.

The image is correctly booting, but none of my wifi boards were setting up.

Then I've tried to set it with the command line, on both air and ground Pis (ip link set _macadress_ up , iw dev _macadress_ set type monitor). The boards were set up in monitor mode, but I had no video on the ground Pi.

I've made this tests with awus036ac, awus036ach and csl300. The awus1900 couldn't be set up, because of the 5.2.20 branch which doesn't support it.

Many thanks for your work.
I think it shouldn't miss a lot before your image works.

Yes21, 2018-09-08:
**@seeul8er**
I also want to test DroneBridge with the modified "russian image".
Could you please explain how to install DroneBridge on an existing wbc image ?

It should be very useful that everyone could test DroneBridge with a modified wbc image.

seeul8er, 2018-09-10:
@Yes21 Just did some tests with my CSL300 and I can confirm that video transmission is not working. It works for about 500 blocks (~3 seconds) and then it stops. Maybe the drivers buffer is filled or something. I'll see what I can do. Maybe moving back to the old driver from the WBC images helps.
Reception seems to be fine with the CSL300.

Regarding the other thing: You need to copy all the files from this repo to a dronebridge folder inside the home folder of WBC. Replace the ".profile" file with the one provided in this repo. Copy the provided "*.ini" files and plugins folder to "/boot". I will not give more detail on that since I do not like the idea of people mixing it all up. In the end we get tons of images like the russian one, where noone knows what the hell they actually changed

Yes21, 2018-09-10:
**@seeul8er**
I've just done more tests with 2 csl300 (one on each side). The video transmission is working and doesn't stop !
With my 8812/8814au, I always see "that device is not up".

I will try tomorrow to follow your advices for the other thing, despite you don't give the end of the process ... I just want to test it, because I'm an old curious man ;)
If I have success I won't share any image, because I'm ok with your wish that noone mix your documented work with undocumented image.

seeul8er, 2018-09-11:
@Yes21 
[I am using these patches](https://github.com/DroneBridge/RPiKernel/tree/master/kernel_4_14_66/patches). The ones you linked are just for documentation and backup. Not all of them work for v4.14.66.

Thanks to your link I think I know what causes the crashes with Ralink. I might forgot to kill the watchdog. I'll try to get a new Kernel running with a fix.

> I've just done more tests with 2 csl300 (one on each side). The video transmission is working and doesn't stop !

Are you referring to my pre-release v0.6?

> I will try tomorrow to follow your advices for the other thing

It's just copy&paste of all files to the right location. Everything goes into a folder `/root/dronebridge`. Inside that folder execute `cmake .` followed by `make` to compile everything.
Then put the config (DroneBridge configs and WBC config) files and plugins folder inside `/boot`. Now you are set (in theory). You can now run `start_db_xx.sh` to start everything. We do that together with some other stuff inside the `.profile` script. You could overwrite the one inside `/root` with the one this git repo provides, but you will lose some of the changes that the russian guy did to it.

Thanks for testing :)

Yes21, 2018-09-11:
> It's just copy&paste of all files to the right location. Everything goes into a folder /root/dronebridge. Inside that folder execute cmake . followed by make to compile everything.

The cmake version of the russian image is to old. I give up !
Will wait for your next image.

Thank you for your advices.

seeul8er, 2018-09-13:
Got a new Kernel with disabled watchdog for RT28xx. And RTS frames as default frames for the video transmission. Seems that RT28 does not like short data frames anymore :(
Tested it very quickly with my Ralink card. Lets see if it works for you.

[Image can be downloaded here!](https://forstudents-my.sharepoint.com/:u:/g/personal/ga25puh_forstudents_onmicrosoft_com/EV3_6oTm8vVMpGCXyhJ_CkIBaZn6Xw4mpa7-GXXgMOq1pg?e=311a4N)

htcohio, 2018-09-14:
@seeul8er what this have fixed the problem with my atheros cards as I shared them with you or would that problem still exists?

seeul8er, 2018-09-15:
@htcohio I doubt it. Right now I do not have any idea why Atheros diversity makes problems. Might be some change hidden in the new kernel code or the decoder software for Broadcom.

careyer, 2018-09-16:
@htcohio : Is it this #30 issue you are talking about? If not: The diversity issue you are talking about ... how does it manifest itself?

Yes21, 2018-09-16:
**@seeul8er**
I've tested your last 0.6 beta image. My results aren't very good : 
- I can confirm that it doesn't boot on a Pi0 (the previous image did boot on a Pi0),
- the CSL300 is working only on the AirPi, not on the ground,
- all the AWUS boards aren't set up in monitor mode. I have to do it with the command line,
- the AWUS1900 doesn't work on both Pis,
- the AWUS036AC/H are working only on the GroundPi.

The only working combination for me was the CSL300 on the AirPi, and AWUS036AC or ACH on the GroundPi, but with very poor video quality and a message with "unknown encapsulation ..."

It seams that's it's no so easy ...

Yes21, 2018-09-22:
**@seeul8er**
I  just succeed in installing DroneBridge 0.6 beta on the "russian image". I had to update cmake, install python3 and a few others python3 library packages.
- 1st test with the russian .profile : the alpha 88xxau boards are working (video transmission ok) and I had to start DB with command line. But I couldn't test if it were really working.
- 2nd test with the db 0.6 beta .profile : the alpha boards aren't set up, and no video transmission. Exactly like with your last 0.6 beta image !
-3rd test after trying to mix the two .profil scripts (!!) : the boards are up in monitor mode, but no video transmission.

My tests perhaps confirm that there is much work to do around the .profile script. But it is so complicated, long and not many documented, that I give up understanding anything ...

Good luck to you ;)

Yes21, 2018-11-02:
Hi Wolfang,
Do you know how I could clone the kernel source 4.14.66 you are using (https://github.com/raspberrypi/linux/tree/c5f7d3c4daf14ba3717fcc1497854d8c365bd742) ?
The `git clone -b ...` command doesn't work, and I don't succeed in finding a working solution on the net.
Thanks if you know how to do.

Yes21, 2018-11-03:
Thank you very much for your help ;-)

```
git clone --depth=100 https://github.com/raspberrypi/linux/
cd linux/
git checkout c5f7d3c4daf14ba3717fcc1497854d8c365bd742
```

dposea, 2018-12-06:
also git submodule update --recursive if the repository has any submodules. Took me weeks to figure out that part.  For example ardupilot has mavlink and nuttx as submodules, and you need the submodule update command to sync those with the version checked out.  Git keeps track of everything so you don't need a version on the submodule update.

TGithubbr, 2019-06-09:
So does this work now or does the issue still persist?