DroneBridge/DroneBridge #34: RPi B3+

Fi156, 2018-09-27:
Hello,

thanks for the great project!

I gave it a try on the new B3+ without a desirable outcome...

Symptoms are described here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=58151
(Yellow text)

I tried with your recent 0.6 image. 
( https://forstudents-my.sharepoint.com/:u:/g/personal/ga25puh_forstudents_onmicrosoft_com/EWi7-HH8Y1pOjLUn_ceffugBhC0Dsemiy5vxi_d9oduu-w?e=yJOoHS )

I will happily test a new version :)
It would be very good, if you provide some info on how to "compile" the repo to an img-file for flashing.

Thanks!

seeul8er, 2018-09-27:
Hi! Thanks for reporting and for providing some info about what could be the cause!
I'll try to get a new test-image running within the next week or so.

Right now the process is:
1. Get the v0.5 image and flash it to a SD card
2. [Get the experimental kernel sources for DroneBridge](https://github.com/DroneBridge/RPiKernel/releases)
3. Modify, compile and install the new Kernel [(applied patches and used configs)](https://github.com/DroneBridge/RPiKernel/tree/master/kernel_4_14_66)

(4. Update the files under `/root/dronebridge` with the ones provided in this repository)
(5. Run `cmake . && make` to compile the DroneBridge modules)

Fi156, 2018-10-02:
I got lost at step 3:
(Especially: "Rename the provided config files for both architectures to .conf depending what kernel you are building." from https://github.com/DroneBridge/RPiKernel ... Cant locate those conf-Files...)

Never mind, I will wait for your new image :)

closed 2020-05-01 by seeul8er