DroneBridge/DroneBridge #35: Source of Android DroneBridge App

redwineproof, 2018-10-05:
As I understand it, Dronebridge is built to be modular, so that users are able to add their custom modules on ground and on air.
To complete the picture, I think it would be interesting to have access to the source of android Dronebridge Application. This way users would be able to add new functionnality from ground station to the drone.

**For example to add a clamp on a drone:**
- control the clamp using dronebridge plugin
- visualize clamp state using dronebridge application

By the way, thank you **Seeul8er** for this great communication system :-)

seeul8er, 2018-10-10:
Hi, thanks for your interest 👍 
At this point I am not willing to share the source code of the Android application. I might add some paid features in the future and release it on the play store. Maybe I'll make the source of the free version open, but not at this point in time.

I can absolutely understand your wish. DroneBridge for Android should support custom messages for that purpose. I will think about it. Should be doable for simple ON/OFF triggers and String messages.
This might overlap in some aspects with #18

closed 2018-10-14 by seeul8er

redwineproof, 2018-10-15:
Thank you for your answer, custom messages should be a good start I guess.

hadogenes, 2019-12-04:
Could you release the source now?
I would like to develop it for ex. VR

seeul8er, 2019-12-05:
Nop sorry. Still not. If you want VR you can go [use this one](https://play.google.com/store/apps/details?id=constantin.fpv_vr.wifibroadcast&hl=de). It does a very good job I think.

There is also a GPL licensed version of it on Github