DroneBridge/DroneBridge #37: raspberry zero w bug

quiotto74, 2019-01-17:
onboard wifi hostpot card not found

system123, 2019-04-07:
The wifi hotspot card should be "wlan0" for the ZeroW running the Raspberian Stretch

quiotto74, 2019-04-18:
unfortunetely this didnt worked so i used an external wifi usb with an usb hub an that worked.

system123, 2019-04-18:
Are you just trying to use it on the ground station as the WiFi hotspot?

quiotto74, 2019-04-20:
yeah for the gnd, 
when i started the  raspberry, it diplayed me the display settings
                                                                        setting up wifi cards
                                                                        pi3 onboard wifi hostpot card not found
                                                                        setting up %first mac% monitor mode
                                                                        warning unsuported or experimental wifi card: rtl8192u
                                                                        setting up %second mac% monitor mode
                                                                        starting rx 
then i  put ifconfig in the terminal and no feedback
so i tried with iwconfig no feedback too

system123, 2019-04-21:
Unfortunately, you not providing enough details to be able to troubleshoot the issue.

Are you running master or nightly branch? 
Have you modified the config file?
Your log says you are using an unsupported wifi card, have you tried removing that or installing the patched drivers for it?
Can you see the onboard RPi wifi before running drone bridge?
If running the nightly build, have you modified the source files so that PI3_NIC is assigned correctly?

Please share your config file, as well as screenshots of ifconfig before running drone bridge as well as after.

closed 2019-04-22 by quiotto74