DroneBridge/DroneBridge #38: Add support for no camera

system123, 2019-03-28:
I have an application where I don't have a camera linked to the pi but rather an audio stream. However, wifibroadcast assumes a camera is connected, it'd be nice to be able to configure this feature and allow for any packet of a specific size to be sent instead of video.

seeul8er, 2019-03-28:
Are you using it on a drone or are you just interested in getting audio from A to B?

It should already be possible and will only require you to change a couple of lines in the script. I can point you to the right location.
I see that adding a configuration for something like this might be interesting for some people, but currently I am pretty packed with my thesis so it won't happen anytime soon.

system123, 2019-03-28:
It will be used with remote beacons that capture audio, and this audio needs to be transmitted to a moving receiver station (boat) with the best effort and low latency - so no need for autopilot etc. Mainly just looking for a way to broadcast audio and some telemetry information, and allow the receiver to capture it without the need for handshakes and connection negotiation.

Would be great if you can point me to the right place in the scripts.

Thanks

pilotnbr1, 2019-03-28:
@system123 As Wolfgang sounds busy (and don’t want to step on his toes) If interested I threw together a branch on Openhd called Enhance-NoCam.

https://github.com/HD-Fpv/Open.HD_Image_Builder/tree/Enhance-NoCam

 I think others might have the same needs. Untested at the moment you would have to build it from the image builder using that branch.

seeul8er, 2019-03-28:
Seems like your options are coming in 😄 
Basically you need two things:

1. Put your wifi adapters in monitor mode
2. Start RX and TXs programs. Data is fed as a continuous stream via a named pipe.

You can modify the v0.5 releases code to fit your needs. Part of the important stuff happens inside `/root/.profile`. It is a big and awful script and will be replaced as soon as possible.

In v0.5 we determine TX or RX side [here. Hardcode it to be 0 for receiver or 1 for the transmitter.](https://github.com/seeul8er/DroneBridge/blob/66ad4597ea1671fc003f5e0906c8dd7d0c5b2257/.profile#L1799)
**To feed your audio stream** instead of the video [change this line](https://github.com/seeul8er/DroneBridge/blob/66ad4597ea1671fc003f5e0906c8dd7d0c5b2257/.profile#L769). Replace the raspivid command with something that outputs your data to stdout.
**To get the data** on your receiving side change [this line](https://github.com/seeul8er/DroneBridge/blob/66ad4597ea1671fc003f5e0906c8dd7d0c5b2257/.profile#L969). You can disable the distribution of the data to different FIFOs and take the received data directly via stdin to your application. Your code would start behind `$VIDEO_BLOCKLENGTH $NICS | <your_app>`
You can also start multiple instances/links of these receivers and transmitters by supplying a different `-p` argument. Basically a virtual port.

Because the script is such a monster I cannot guarantee that all of the rest will be stable. Sometimes it [checks for raspivid to be running before it does stuff](https://github.com/seeul8er/DroneBridge/blob/66ad4597ea1671fc003f5e0906c8dd7d0c5b2257/.profile#L1774). So that won't work since we disabled it.

Alternatively you can have a look at the nightly branch. [I wrote a python script that puts the cards into monitor mode](https://github.com/seeul8er/DroneBridge/blob/nightly/startup/init_wifi.py). This should work. (Just like any other method for setting cards to monitor mode)

After that you can launch the TX and RX programs. You can use the ones from v0.5 or a [newer](https://github.com/seeul8er/DroneBridge/tree/nightly/video) (not so much tested) version. [If you start the transmitter with the `-a 1`](https://github.com/seeul8er/DroneBridge/blob/3b8cafb77a6e5daa51e3c108d6af38f3ddfb78d4/video/video_main_air.c#L204) option it should also work with Ubuntu and non patched drivers.
That way you could kind of disable the entire `.profile` script and concentrate on the stuff you need.

Hope this gives you some insite. Just ask if something is unclear. I have tested the newer transmitters with image files instead of video. That worked fine. Hope it does well with audio.

BTW. [This might be usefull as well](https://github.com/svpcom/wifibroadcast). It is an alternative transmitter app that basically does the same as DroneBridge but seems to accept UDP streams as a source.

system123, 2019-03-29:
@seeul8er @pilotnbr1 Thanks for the detailed instructions and help, I think I should manage to achieve what I need from here. Will reopen the issue if I get stuck.

closed 2019-03-29 by system123