DroneBridge/DroneBridge #41: Discontinue iNav support

seeul8er, 2019-05-29:
Supporting iNav & MAVLink is very time consuming for a single developer and makes the system more complicated since iNav "needs" two serial ports (MSP & LTM)

Looks like Ardupilot now supports cheaper hardware, so the big advantage of iNav over price seems to be gone now. Does it make any sense to keep supporting iNav? Ardupilot looks like the go-to option for me:

- Better library support, SITL Simulator
- Better documentation
- More features
- Bigger community

mickey2011, 2019-06-09:
But the Ardupilot GUI is a mess. The INAV GUI is a lot cleaner these days

TGithubbr, 2019-06-09:
I think iNav is pretty irrelevant these days. Maybe make it like a module and try to carry it on to newer versions and don't implement features for it which would require changes to the core iNav communication. Or maybe someone in the community could do it?

closed 2019-11-26 by seeul8er