DroneBridge/DroneBridge #43: Wifi not recognize by RX but TX does

diyquad, 2019-07-13:
Hi,
I'm trying to setup dronebridge.
I use 2 wifi adapter: alpha S036NHA and alpha S036NEH
Both card are well recognize on the TX. (card found, card ready message is displaying!)
But when i try with the RX, i have a message
**"PI3 Onboard Wifi Hotspot card not found"**

Dronebridge continue loading, I still get a picture, but very bad, frame are partial and not usable. Also the Dbm always in red even if card are 1m of each others 

I try to change to RPI foor RX, i tryed a PI0, PI2 and PI3 always the same (with reflash dronebridge when change). 
Strange also message is always "Pi3 on board..." even with a pi0.

No matter what the RX never recognize the wifi card, that TX does.

**Software/Configuration:**
Drone Bridge 0.5

**Hardware**
PI0 as TX
PI0 as RX (also PI2 or PI3)
Alpha AWUS036NHA 
Alpha AWUS036NEH
Picamera v1
5v-2A power for each

Thanks for your help

seeul8er, 2019-07-15:
Try disabling the WiFi hotspot and see if something changes. You can do that by setting `WIFI_HOTSPOT=N` inside the Wifibroadcast config file.

Which of the cards is used on GND and AIR side?

diyquad, 2019-07-15:
Thanks for your help

When disabling wifi hotpost, i only have the following message
```
**Setting up wifi cards....
Setting up 00C0ca ....................**

```

For the card, i'm switching, as i'm still testing and it's not working. Both card are OK on TX but not on RX.
(on TX i have a
```
**waiting for wifi card to become ready.... card ready    
setting up wifi card .... 
setting up 00C.....**."
```

seeul8er, 2019-07-22:
Strange... At the moment I am working on getting a new release out the door that will do a lot of things quiet differently when it comes to setting up the cards. Because of that and because I have no idea on what could cause your issues I will close this for now.

You could try using the pure WifiBroadcast RC1.6 release and see if that works for you. That release is very robust. If the issue continues to exist it's most likely a problem with your power supply or something.

Sorry I can not be of more help at the moment. Hope the new DB0.6 release will fix that issue.

closed 2019-07-22 by seeul8er