DroneBridge/DroneBridge #45: Compiling on Raspbian

Summer88, 2019-11-25:
Trying to build the last version of DB on a pizero with 2019-09-26-raspbian-buster-lite.

I've followed:
https://dronebridge.gitbook.io/docs/developer-guide/downloading-and-compiling

But getting stuck here:

> pi@rover:~/DroneBridge $ cmake .
> -- control module: Release configuration
> -- db_common module: Release configuration
> -- status_ground module: Release configuration
> -- db_common module: Release configuration
> -- proxy_ground module: Release configuration
> -- db_common module: Release configuration
> -- telemetry module: Release configuration
> -- db_common module: Release configuration
> -- video module: Release configuration
> -- db_common module: Release configuration
> -- usbbridge module: Release configuration
> -- Checking for module 'libusb-1.0'
> --   Found libusb-1.0, version 1.0.22
> -- db_common module: Release configuration
> -- syslog_server module: Release configuration
> -- db_common module: Release configuration
> -- db_common module: Release configuration
> -- Could NOT find PCAP (missing: PCAP_LIBRARY PCAP_INCLUDE_DIR) 
> -- Performing Test PCAP_LINKS_SOLO
> -- Performing Test PCAP_LINKS_SOLO - Success
> -- Looking for pcap_get_pfring_id
> -- Looking for pcap_get_pfring_id - not found
> -- osd module: Release configuration
> -- db_common module: Release configuration
> -- db_common module: Release configuration
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> PCAP_LIBRARY (ADVANCED)
>     linked by target "rx_vid" in directory /home/pi/DroneBridge/video/legacy
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/pi/DroneBridge/CMakeFiles/CMakeOutput.log
[CMakeOutput.log](https://github.com/seeul8er/DroneBridge/files/3887145/CMakeOutput.log)
".
> See also "/home/pi/DroneBridge/CMakeFiles/CMakeError.log
[CMakeError.log](https://github.com/seeul8er/DroneBridge/files/3887150/CMakeError.log)
".
>

seeul8er, 2019-11-25:
Hi, I think you are missing the libpcap library.

`sudo apt-get install libpcap-dev` should get you running.

I'll see if I can get you my latest development image. Should already run on the Pi0

Summer88, 2019-11-26:
Thank you for fast reply, I will retry.

May  I suggest a tool like this:
https://github.com/HD-Fpv/Open.HD_Image_Builder

I'm looking at different projects at the same time, I didn't get why there are so many of those, it wouldn't be better to make a single project and all people working on that?

Thanks, BR

seeul8er, 2019-11-26:
Although it seems to work for them I am not a fan of the image builder. Once the base image for the Raspberry is done you do not really need to change it much during development. So why spend all that effort on developing a stable image builder in the first place. Users will always just download the ready made release. Different philosophies, concepts and demands drive the projects in different directions.

BTW: You can also use an image created by the linked tool and then go install/clone DroneBridge on that. That might get you up and running the quickest. You might need to adjust some paths if you want to use the start scripts or the communication module of DroneBridge.

closed 2019-12-02 by seeul8er