DroneBridge/DroneBridge #48: v0.6 login at startup

diyquad, 2020-01-04:
Hi,
I'm installing dronebridge last version (0.6) on RPI3 (but i test also on PI2 and Pi0 Wh)

I install with the IMG you provide.

At first boot, it is asking for "dronebridge login"
tryed several combinaison, but nothing work

what are the access (login/password?)

I notice after a certain time that dronebridge is still loaded on top (overlay) of the ssh login screen

thanks

seeul8er, 2020-01-04:
The [login info can be found here](https://dronebridge.gitbook.io/docs/developer-guide/downloading-and-compiling#default-ssh-login-data)

The overlay is the OSD I guess. To kill it you need to stop the service startdb. See the Docs on more detailed info

diyquad, 2020-01-04:
already tryed the pi/raspberry combinaison, doesn't work;
but will try again and let you know

thanks

closed 2020-01-05 by diyquad