DroneBridge/DroneBridge #49: high-speed data transmission via UART (no need to transfer images)

marseric, 2020-01-06:
This is really a good project, thank you author.
Please tell me, I have two Raspberry Pi 3Bs. 
I only need to use it for high-speed data (1 Mbps or higher will be great )transmission via UART (no need to transfer images). 
How should I configure it? Thanks a lot

seeul8er, 2020-01-06:
Set the data rates as low as 2, 5 or 6 Mbit/s on the air side. I guess you will want to use FEC for your data, so you will need to use the video module. 

**My suggestion:**
You will not need most of the DroneBridge modules so you can disable them.
You can do that by disabling the [service](https://dronebridge.gitbook.io/docs/developer-guide/startup-and-logs). 
Running `sudo systemctl disable start_db` should do the trick. 

Because the service is disabled you will need to run the Wifi-Adapter setup script manually on boot. To do that call `python3.7 /home/pi/DroneBridge/startup/init_wifi.py` on the air side and `python3.7 /home/pi/DroneBridge/startup/init_wifi.py -g` on the ground side. After that you can run the data transmission.

[Here is some more detailed info on how to use the video module.](https://dronebridge.gitbook.io/docs/dronebridge-for-raspberry-pi/standard-modules) On the air-side something like

`sudo cat myUARTdatasource | video_air -n wifiinterface -n -c 200 -t 2` 

should read & send. 

To receive call `sudo video_gnd -n wifiinterface -c 200 -u N | tee output.txt` to write to a text file. Adjust this to your needs

seeul8er, 2020-01-06:
If the data rates are low enough some of the cheaper LoRa ESP32 modules could be a better choice. They are less complex, cheaper and will probably give you more range. Not sure if there is a ready to use firmware for your purpose.

marseric, 2020-01-07:
Thanks, you are so kind. I will follow your instructions. Thank you again

marseric, 2020-01-07:
I have tested your esp32 project and it works very well,  but its UART speed is limit to 115200bps, so. .

closed 2020-05-01 by seeul8er