DroneBridge/DroneBridge #51: Release Image File too large for 4GB micro-SD card

course-up, 2020-04-19:
**Short Description**
The provided Alpha 0.6.0 img File is too large for most micro-SD cards.

**Steps to reproduce the behavior:**
1. Goto Releases Page https://github.com/DroneBridge/DroneBridge/releases
2. Click on the provided link "Get the ready to use image for the Raspberry Pi here"
3. Unpack DroneBridge_0_6_0.zip file 
4. The DroneBridge_0_6_0_alpha.img has the size 15.931.539.456 bytes (~14.8GB)

Advise on howto write to 4B or 8GB mirco-SD cards would be great. The wiki mentions that a 4GB micro-SD card is sufficient.

seeul8er, 2020-04-19:
It involves a couple of extra steps to shrink the images. For development purposes, it is also more convenient to have a bigger partition. Since this is an alpha release I will not fix this for now. I do also think that it is no big issue since it got kind of hard to find SD cards <16GB.

closed 2020-04-19 by seeul8er

course-up, 2020-04-19:
Thank you for the quick reply. It might be worth mentioning/changing in the wiki or the release page, it was a little bit misleading.