DroneBridge/DroneBridge #53: Will this program be supported on openwrt in the future

lsy512, 2020-04-28:
This is a good program.openwrt is a specialized routing system.Will this program be supported on openwrt in the future?

Dronotique, 2020-04-28:
To follow... I vote also for this

seeul8er, 2020-04-28:
OpenWRT is for routers etc. What use case would there be for DroneBridge? I barely have time to update the Android App for now. Supporting OpenWRT would be all the way down on my personal priority list. Sorry for that.

The software-based on the DB common lib should compile fine. It only uses very basic low level kernel functions like raw sockets and BPF.

lsy512, 2020-04-29:
The router comes with WiFi, so many router modules can add a power amplifier to realize these functions

Dronotique, 2020-04-29:
Routers also have RJ45 plugs. 

That could allow to have UAV <-> Wifi <-> OpenWRT <-> MultipleRJ45 <-> MultipleGCS

seeul8er, 2020-05-01:
I see. Routers would make for a nice dedicated GND station. I agree with that. It would be worth giving a shot. 
However, I personally do not possess a spare OpenWrt capable router, so do not expect any development from my side right now.

Dronotique, 2020-05-01:
I can advice the Xiaomi Mi Router 3G that give a good value for money

seeul8er, 2020-05-01:
It ain't that easy. The router must have a wireless chip that supports monitor mode and packet injection. The packet injection rate should be rather high. That probably leaves us with routers that have simular chips that the currently supported wifi adapters.

Dronotique, 2020-05-01:
OK. How I can know if the router wifi chip supports monitor mode and packet injection? Is there a linux command?

seeul8er, 2020-05-01:
By searching the internet. Aircrack & WikiDevi offer some databases and checks on monitor mode and injection. It is all about the driver. OpenWRT also has to support that specific driver. So as you see ... no plug & play here :D

closed 2020-09-16 by seeul8er