DroneBridge/DroneBridge #54: no .ini files found on SD-card found after flashing dronebridge 0.6.0

bpjdons, 2020-05-03:
Reporters can deviate from this template if they have a clear understanding of the problem and the software parts that might be involved.

**Describe the bug**
A clear and concise description of what the bug is. Does/Did it happen on earlier releases?

**(if useful) Steps to reproduce the behavior:**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software/Configuration:**
 - [DroneBridge release version]
 - [if used: Android app version, Smartphone type and OS version]
 - [DroneBridgeAir.ini from the AirPi] not found
 - [DroneBridgeGround.ini from the GroundPi] not found

**Hardware**
 - [Raspberry Pi 3 etc.]
 - [WiFi adapter on ground station]
 - [WiFi adapter on UAV]
 - [Any other connected devices]

**Additional context**
Add any other context about the problem here.

seeul8er, 2020-05-03:
[Seems like a mounting and resizing problem.](https://gitter.im/DroneBridge/Lobby). Seems like some have already solved it. Maybe they can give a hint. I'll see when I find some time looking into it.

bpjdons, 2020-05-11:
Hi Wolfgang,

 

Did you find a solution for this no .ini files not found?

seeul8er, 2020-05-12:
Hi, sorry so much other stuff that I need to finish first.
Try the following: Identify the DroneBridge partition that is not mounting. It should be around 13GB in size. Format it to exFAT. Copy the config files etc. to the formatted partition (see attached files).

Could not test it but maybe that works.
[DB.zip](https://github.com/DroneBridge/DroneBridge/files/4618593/DB.zip)

bpjdons, 2020-05-13:
Hi Wolfgang,

 

The program reports that there was no Wifi adapter found. I am using the Alpha longrange adapter.

This adapter works with Raspberri Pi 3b.

 

Also tried the TP-Link, that one was not recognized either (likewise with the PI 3b.

 

Thanks,

bpjdons, 2020-05-18:
It works. I can see the program starting up till the point where it asked to login.

 

I have connected my Wifi adapters and configured the Rasperri PIs 3b.

 

I can not connect to the Wifi hotspot. (the same configuration works with Ez-wifibroadcast).

 

Please advise.

 

Thanks,

Bram

seeul8er, 2020-05-19:
That is strange. Are you still on the stock image (except for the file system format maybe)? I cannot help you if you do not provide logs, but only a 'it does not work'.

Can you post the log you get by typing `journalctl -b` after you tried to connect to the access point?

bpjdons, 2020-05-19:
OK, this is the situation now:

 

I do have a video connection with the PI groundstation (which is connected to the HDMI port).

 

At the point where the login prompt is shown I  tried to login with “pi’’ and ‘’raspberri’’.

The response is ‘’access denied”.

 

If you press ‘’alt + f10”’, you normally (with ez-wifibroadcast) gets you the setup screen in which you can enable ssh.

 

So I can’t sent you a log file. Still no access from my tablet via wifi to the ground PI.

 

Regards,

Bram

seeul8er, 2020-05-19:
In DroneBridge there is only one screen. No need for ALT+Something.
Try to login using `raspberry`

bpjdons, 2020-05-19:
Sorry, does not work either.

 

After displaying the login, it is difficult to login. The program most times just continues.

seeul8er, 2020-05-20:
Connect the Pi via ethernet und just use SSH

bpjdons, 2020-05-20:
I can not connect via ethernet because the PI did not obtain an IP address from my DHCP server.

 

With ez-wifibroadcast it gets an IP address of 192.168.178.47.

mahmut20, 2021-03-24:
> Hi Wolfgang, Did you find a solution for this no .ini files not found?

maybe i can help you last time i got a issue that .ini did not apply. but i was changing wrong file.
i have a question which file are you trying to change?

closed 2021-07-16 by seeul8er