DroneBridge/DroneBridge #55: Pi 4 Compile error! fontinfo.h : No such file or directory

Eronwu, 2020-06-02:
when I compile this project ON MY PI 4 as wiki said:
git clone https://github.com/seeul8er/DroneBridge.git
cd DroneBridge
# git checkout nightly
git submodule init
git submodule update
cmake .
make

I got a error on the screen:
Building C object splash/CMakeFiles/db_splash.dir/main.c.o
/DroneBridge/splash/main.c:26:10: fatal error: fontinfo.h : No such file or directory
#include "fontinfo.h"
compilation terminated.

Does anybody occurred this error ??? PLS help!
Did this project lack of some code or Pi 4 lack of some library? 
But this head file as I know, is just a font usage file, maybe used in OSD module.

seeul8er, 2020-06-02:
As far as I know the Pi4 does not support OpenVG. However, the OSD & the splash screen require that dependency. You can compile and run the other modules by commenting [lines 32 & 33 the the CMakeLists.txt.](https://github.com/DroneBridge/DroneBridge/blob/master/CMakeLists.txt#L32)

Eronwu, 2020-06-18:
> As far as I know the Pi4 does not support OpenVG. However, the OSD & the splash screen require that dependency. You can compile and run the other modules by commenting [lines 32 & 33 the the CMakeLists.txt.](https://github.com/DroneBridge/DroneBridge/blob/master/CMakeLists.txt#L32)

but when execute the start_db, failed again.. 
seems that the code has deep relate to the osd module...
`/etc/init.d/start_db: line 33: ./osd: No such file or directory`