DroneBridge/DroneBridge #56: Cant get default raspberry pi image to work

Freshdouble, 2020-06-26:
I don't know if its a bug, or i'm using the image wrong.

But my problem is, that i can't get the data link to work between two raspberry pis.

I'm using a Raspberry Pi Zero with a Atheros based wlan stick and the raspberry pi camera as the air pi and a Raspberry Pi 3 with the same wlan stick as the groundstation.
When i plug in the groundstation to my monitor, it boots up and shows the osd in front of the text console but nothing else happens. The wlan stick starts to blink shortly and then shows no activity. On the air pi, the wlan stick shows the same behavior, but the light on the camera is turned on, indicating it is active.

I'm using the 0.6Beta Image from the downloads and didn't modify the config files on both pis. The only action i took was to run make in the home directory of the pis. I just want to use the video stream of dronebridge the Getting Started Page of the wiki didn't help and i assumed, that the two raspberry pis should run out of the box with the provided image.

I tested the mode of the wlan interface with iwconfig and it shows that the stick is in monitor mode, as it should be however the interface name is somehow strange.

Did i miss anything? Do i have to make specific settings to make the video link work?

seeul8er, 2020-06-27:
The Raspberry Pi Zero is known to have startup issues with the current DroneBridge image. I do not have the hardware at hand to debug the issue. But since people still keep buying the Pi0, despite its low performance and pain for debugging, I ordered one for me yesterday. However, I will be on vacation so do not expect any test results within the next two weeks.

The image should work out of the box. The fact that the camera LED is on is a bit confusing, I'd expect it to be off. You could try using your Pi3 as the AIR Pi. The Pi0 will probably not be able to handle the video decoding (FEC) + OSD properly on the GND, but you should get some kind of crappy image via HDMI.
If you need a fast fix - try replacing the Pi0 with a different one. The Pi2 and Pi3B are reported to be working. I did not test the Pi3A+, but it should also make for a nice AIR unit.

Freshdouble, 2020-06-28:
Thank you for the quick response. I will try it with 2 Pi 3Bs.

seeul8er, 2020-10-01:
New release.

closed 2020-10-01 by seeul8er