DroneBridge/DroneBridge #57: Nvidia Jetson Nano with 4K Camera

Zipdox, 2020-07-29:
Would it be possible to port this project to the Nvidia Jetson Nano and use it with an Arducam High Quality Camera to achieve 4K video recording?

seeul8er, 2020-07-29:
Shure! If you want to use the Jetson on the drone you will need to replace the "[raspivid](https://github.com/DroneBridge/DroneBridge/blob/master/startup/start_db_modules.py#L209)" call with something that outputs a video stream via the stdout. The command could look something like this:
`gstreamer --super_cool_arg --h264_pls > sudo ./video_air -n wireless_adapter_in_monitor_mode -c 200 -t 2`

On the receiving side, [you will need something to decode and display the video](https://github.com/DroneBridge/DroneBridge/blob/master/startup/start_db_modules.py#L118). The current command accepts a raw H.264 stream. We use the hello_video.bin app that uses "native" RPi libs and only runs on those boards.

You do not need to patch the kernel. Only if you need more tx power etc. [You can also just run the video module on AIR and GND for testing send & receiving video.](https://dronebridge.gitbook.io/docs/dronebridge-for-raspberry-pi/standard-modules#video-module) No need to get all the other modules.

Zipdox, 2020-07-30:
I'm just brainstorming about possible projects while I'm on vacation, I don't have a Jetson Nano yet. I'll see what I'll do when I get back.

seeul8er, 2020-07-31:
I already added a comment that the Jetson should work. In fact DB is not a Raspberry Pi only project. It can in theory run on all linux systems. Closing this for now since not an issue

closed 2020-07-31 by seeul8er

jptalledo10, 2020-09-27:
@Zipdox 

Just wondering which configuration/settings did you use for gstreamer to send 4K video along Dronebridge?

Zipdox, 2020-09-30:
I moved on to other projects. I don't have to budget to mess around with expensive boards like the Jetson and the camera is also quite pricey.