DroneBridge/DroneBridge #58: DroneBridge for Android : Beta is expired!

trogalko, 2020-08-04:
Hi, today I tried to launch DroneBridge for Android V0.6 Beta, but got Beta is expired! message.
Is it time limited application? Could You please kindly release new version? 
Thank You so much.

seeul8er, 2020-08-05:
It's limited so users are forced to update the application ~every 6 months.
I'll see that I release one with an updated timeframe today

seeul8er, 2020-08-06:
[Can you check the release with updated period please](https://forstudents-my.sharepoint.com/:u:/g/personal/ga25puh_forstudents_onmicrosoft_com/EdzSWxE7ZMhIvGesmLFodUABrMjW3oSqpn6d4RU_TgZ3JQ?e=TDdAnH). USB connection support still broken, but I am working on it already

trogalko, 2020-08-07:
> [Can you check the release with updated period please](https://forstudents-my.sharepoint.com/:u:/g/personal/ga25puh_forstudents_onmicrosoft_com/EdzSWxE7ZMhIvGesmLFodUABrMjW3oSqpn6d4RU_TgZ3JQ?e=TDdAnH). USB connection support still broken, but I am working on it already

Yes, It works, thank You

closed 2020-08-07 by trogalko