DroneBridge/DroneBridge #61: Realtek  RTL8812AU Driver not working on DroneBridge 0.6 Beta image

jptalledo10, 2020-09-27:
I tried to compile from source the Driver but I believe the kernel had been modified I am getting the following error:

[ 1443.513238] 88XXau: disagrees about version of symbol module_layout.

Running modprobe to the driver on the image. it shows:
[ 2115.054551] usbcore: registered new interface driver rtl8xxxu

But it seems it is not detecting the WiFi card.

Any suggestions to get this fixed?

seeul8er, 2020-09-27:
I am aware of the issue. I am compiling a new kernel as we speak to get release 0.7 out. DroneBridge for Pi uses a custom kernel with patched drivers and firmware

jptalledo10, 2020-09-27:
@seeul8er 

Thanks for addressing this problem. Please advise when/where I will be able to access the release 0.7

Appreciate your support and help in advance

seeul8er, 2020-10-01:
I uploaded a new release for testing.

closed 2021-07-16 by seeul8er