DroneBridge/DroneBridge #68: using rc values in plugins

hex0r, 2021-02-26:
Hi all,

First of all awesome project , thanks for sharing 

I try to use the project for a rover, without any flight controller 
So i wrote a plugin to get the rc values to control the rover with odrives

If some interested, this is my python version of the rc example plugin 
```

import numpy as np
import mmap
from posix_ipc import SharedMemory
from ctypes import sizeof, memmove, addressof, create_string_buffer
from time import sleep
from ctypes import Structure, c_uint16, Array

rc_region = SharedMemory("db_rc_values_t")

while True:
    shm_buf =mmap.mmap(rc_region.fd, 2*14)
    msg_bytes = shm_buf.read()
    rc_dt = np.dtype([('ch', np.uint16)])
    buffer = np.frombuffer(np.array(msg_bytes), dtype=rc_dt)
    for ch in buffer:
        print(ch, end='')
    print("")
    sleep(0.5)

```
additional i have removed the write_to_serial int the DB_RC_PORT section of the control_main_air.c which have some blocking behavior due to there is not FC connected (quick and dirty)

But my question is:
if i unplug the joystick or lose connection, the RC values in the shared memory remain unchanged.
How could i set them to zero or something else if connection get lost to turn off the motors of the rover 
or is there an other option to check if RC is working inside a plugin?

thx for your help

seeul8er, 2021-02-27:
Hi! Since you are the first one really using this feature I never really thought about that problem. You are right. At the moment there is no way to check for a plugin if the data is up to date.

I'd propose to add the timestamp [to the shared memory structure](https://github.com/DroneBridge/DroneBridge/blob/master/common/shared_memory.h#L27) ([add timestamp here](https://github.com/DroneBridge/DroneBridge/blob/master/control/rc_air.c#L267)). That way a plugin can check if the data was updated recently.
This should be a fairly quick fix. Let me know if that would work for you. I'll push an updated to the git afterwards.

seeul8er, 2021-02-27:
Alternatively I could add a check to the control module. I have a 1Hz loop running there for sending status updates. If the RC values are too old I'd set them to 5000 or something. That would add a delay of 1s. Not sure about that

hex0r, 2021-02-28:
Hi seeul8er

Thanks for your help!!

I added the suggested changes to my fork https://github.com/hex0r/DroneBridge
maybe you can check it 
the shared memory looks now,
```
typedef struct {
    uint16_t ch[NUM_CHANNELS];
    uint16_t rssi;
    uint16_t recv_pack_sec;
    struct timeval last_update;
} __attribute__((packed)) db_rc_values_t;
```
I add also a python example to the plugins folder for demonstrating.

thx Arnold