DroneBridge/DroneBridge #69: Is this project no longer under maintenance?

cgrrty, 2021-03-24:
**Feature requests that only target a small user group will be ignored. Every UAV project is different. The developers are not your personal programmers. DroneBridge can easily be personalized using plugins. Those are powerful, can be used with every future release and are easy to develop. Please read more about it in the Wiki.**

e.g. Custom actions that can be triggered via a RC hardware switch are highly individual. Please consider a plugin or a new kind of communication message. The Android app is considered to be the main user interface.

Support for new hardware depends on its benefits and availability.

**Is your feature request related to a problem? Please describe. Make clear why many users would profit from the new feature.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered (if you did)**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

CopterGUI, 2021-03-24:
Seems to be the toy-project of a single person.

If you are looking for something similar that works, use OpenHD, it's much better.

cgrrty, 2021-03-24:
I think this is more standardized than openhd's code, and it's more suitable for people who want to learn programming.

CopterGUI, 2021-03-24:
Why do you think that?
Because you read it in the dronekit wiki ?

seeul8er, 2021-03-24:
@CopterGUI please do not promote here. Seems like there is still enough interesting content to learn from to keep lingering around the repo.
@cgrrty I am the only active developer right now. Development is halted since I am trying to get my 3D printed drone hardware ready. I spent so much time developing DroneBridge that I never managed getting the hardware running. That will still take a few weeks, after that I will fix some issues here.
I do not know anything about the latest releases of OpenHD. Their docs are still lacking a lot. Looking at the lastest code it looks like they finally rewrote much of it. This is just as much "toy-project" as OpenHD. In fact much of the new stuff that OpenHD is doing now (app, library, docs, rewrite of WBC code, logo etc.) was done with DroneBridge 3 years ago. In my opinion DroneBridge is still cleaner and more modular. The docs are also better. Latest hardware might be better supported by OpenHD. I'd argue that none of the projects are simple and thus easy to understand for beginners.

CopterGUI, 2021-03-25:
Hello @seeul8er 
This was not not meant as promotion. I'm just a drone enthusiast and user of all of these project's. 
Not a developer. 
What made me angry is that your wiki is very misleading to new users. It hard to gather information as a newcomer and people will believe what they read there.
And I'm not lingering around your repo, that's the new GitHub app that sends me tons of notifications, I try to turn it off and will not disturb you any longer.

Anyway my choice of words was unnecessary harsh and I want to apologize for that.

cgrrty, 2021-03-26:
@seeul8er thank you

closed 2021-03-26 by cgrrty