DroneBridge/DroneBridge #73: DroneBridge 07. Beta No AP present

bartstech, 2021-06-29:
Using Pi 3B+, stock 0.7 Beta DroneBridge Pi Image and internal Wi-Fi chip

There seems to be zero AP broadcast of DroneBridge. I've ssh'd in to the pi, validated the DroneBridgeConfig.ini parameters:
wifi_ap=Y
wifi_ap_if=internal

No matter what I mess with, I see zero AP broadcast. I've tried changing the channel and went in to apconfig as well to see if anything was wrong. At this point, I do not know why I'm not seeing any broadcast.

seeul8er, 2021-06-29:
What are you using to check? The app? You mean the video is missing? Do you see output via HDMI?
The Pi0 is no good as a GND station, use the Pi3 for the GND.

bartstech, 2021-06-29:
I'm using both my android phone, the app and laptop for the AP broadcast. The Pi I'm using is a Pi3b+ for the GND.

bartstech, 2021-06-29:
I should clarify that I do not see "DroneBridge" AP.