DroneBridge/DroneBridge #76: Injection test failed

cgrrty, 2021-07-29:
https://github.com/DroneBridge/DroneBridge/tree/master/InjectionTest
I did a test on my desktop. 
Testing process. 
2 x rlt8812au
1 x pc ubuntu 18.04

receive.sh
```
MY_WIFI_CHANNEL=149 #5ghz channel
# MY_WIFI_CHANNEL=13 #2.4ghz channel

MY_RX="wlxbcec238a23a5"
#MY_RX="wlx08107bbfcf00"

sudo rfkill unblock wifi

sudo ifconfig $MY_RX down
sudo iw dev $MY_RX set monitor otherbss fcsfail
sudo ifconfig $MY_RX up
sudo iwconfig $MY_RX channel $MY_WIFI_CHANNEL

sudo ./receive  $MY_RX

```
send.sh
```
MY_WIFI_CHANNEL=149 #5ghz channel
# MY_WIFI_CHANNEL=13 #2.4ghz channel

#MY_RX="wlxbcec238a23a5"
MY_RX="wlx08107bbfcf00"

sudo rfkill unblock wifi

sudo ifconfig $MY_RX down
sudo iw dev $MY_RX set monitor otherbss fcsfail
sudo ifconfig $MY_RX up
sudo iwconfig $MY_RX channel $MY_WIFI_CHANNEL

sudo ./send  $MY_RX

```

Test result
The receiver cannot receive data.

seeul8er, 2021-08-08:
Damn. Looks like the rlt8812au drivers still suck. What Wifi Hardware are you using? 
To be honest, I can not really recommend rlt8812au/rlt8814au devices so far. I only got a rlt8814au to test and that one sucks.

cgrrty, 2021-08-09:
I used this driver
https://github.com/aircrack-ng/rtl8812au.git
My wifi hardware works well on openhd systems.
[https://gd4.alicdn.com/imgextra/i3/2085915688/O1CN01SV9RMS1rt9IopnII0_!!2085915688.jpg](url)

Sgw32, 2022-11-10:
Injection test from repository uses different COMMID (communication IDs) and also some other parameters are also different. Use "communication" folder examples instead. However I also still have issues with it :)