DroneBridge/DroneBridge #78: SharePoint links broken

parri0923, 2021-11-08:
Links present in Read Me document for both Pi Image and Android APK both broken.

SharePoint returns the message:

This link has been deactivated.
Sorry, access via this link has been removed by administrator policy. Reach out to the person who shared it with you.
TECHNICAL DETAILS

Troubleshoot Microsoft SharePoint Foundation.
Correlation ID: 773101a0-9060-3000-64b1-e68bb274fe54

Date and time: 11/08/2021 10:58:35 AM

seeul8er, 2021-11-15:
Updated

closed 2021-11-15 by seeul8er

parri0923, 2021-11-16:
Apologies, but the link for the Dronebridge Android APK is the 2.0.2 beta version. It provides the message "The beta version is expired!" and is unusable.

parri0923, 2021-11-17:
The Pi Image is also the 0.7 beta version, not alpha.

seeul8er, 2021-11-22:
Sorry about that. I'll re-upload to something proper and try to fix the app.

seeul8er, 2021-11-22:
RPi image updated again. The app will follow

parri0923, 2021-11-22:
Thanks man!