DroneBridge/DroneBridge #81: How can I use this as a standalone program on Ubuntu?

Jcwscience, 2021-11-26:
I have an Nvidia arm64 board that I would like to use, but I’m having considerable difficulty getting the program to run. I’ve changed all of the file paths I can find in the project source to match my install directory, but it seems like some of the scripts are dependent on the file system layout of a raspberry pi. Is this the case or am I not using the source built correctly? Thanks!

seeul8er, 2021-11-27:
Yes, you may need to adjust paths in the startup scripts folder. The video capture and display part are also RPi specific. You will need to exchange the calls to raspivid and hello_video (video player on the receiving side) with valid calls to something supported by Nvidia.
You should try starting and testing the modules one by one. Manually run the `init_wifi.py` script to enable monitor mode. Then manually start the control module and always listen for errors etc.

Jcwscience, 2021-11-27:
Ok, got it. Do the instructions on the wiki for enabling monitor mode apply to the base raspberry pi image? For the default frequency, or do I need to edit the air pi config?

seeul8er, 2021-11-27:
[This is a generic linux script that should work](https://dronebridge.gitbook.io/docs/dronebridge-desktop/enabling-monitor-mode) That way you do not need the config files, just enter the values inside the script directly. Enabling monitor mode is nothing special to DroneBridge. It's always the same. DroneBridge just makes sure we are locked to a channel and data rate

Jcwscience, 2021-11-27:
Right, I have used that script. I just wanted to make sure I don’t need to change anything on the air pi first. I’ll see if I can get the control module started. I’m running a bit of a hackish Ubuntu version to start with, but I think I can get it working. Thanks again for the help!

Jcwscience, 2021-11-29:
@seeul8er I have got monitor mode working, and I have adapted the script but I keep getting this message endlessly scrolling through the terminal.

b rc values memory open: Waiting for init : permission denied.

Any idea what is causing that?

seeul8er, 2021-11-30:
That is a call to open a shared memory segment under [/db_rc_values_t](https://github.com/DroneBridge/DroneBridge/blob/84bc61ee3f97604d60e2da2e71047f1277f2b553/common/shared_memory.c#L108). It is not required for a basic setup. It allows to overwrite RC values by other applications.

Run the application as root (required anyways to be able to send using monitor mode). If that does not fix the problem you must change the path in the code to something where you have write rights. `/var/db_rc_values_t` or `/home/yourname/db_rc_values_t` should do the trick.

Jcwscience, 2021-11-30:
Ok now we’re getting somewhere! One more question, how do I access the telemetry feed from something like apm planner? Is it on a local port?

seeul8er, 2021-11-30:
You can [receive telemetry via the proxy module on TCP port: 5760](https://dronebridge.gitbook.io/docs/developer-guide/system-architecture#gcs-connected-via-network) - right click on the graphic to open in a new tab - then you can zoom in a bit so the text of the SVG gets rendered correctly. Telemetry is a bit untested in the latest releases.