DroneBridge/DroneBridge #84: h264 usb camera stream

hex0r, 2022-01-07:
Hi all, 

At the moment i test different usb cameras with DroneBridge.
My problem is,  if i use USB cameras with native h264 support i can stream the video over network for testing using the following commands 
  ## Tx
  gst-launch-1.0 v4l2src device=/dev/video2 ! video/x-h264,width=1280,height=720,framerate=30/1 ! rtph264pay ! udpsink host=xx.xxx... port=5600
  ## Rx
  gst-launch-1.0 udpsrc port=5600 ! "application/x-rtp, payload=96" ! rtph264depay ! decodebin ! autovideosink sync=false

works perfect!

if i try to send the steam via dronebridge I get a clear video stream as long the camera is not moving.
but if i move camera it gets very noisy, and latency is also bad 
for me, it looks like some encode decoder issue 
maybe someone has a idea on this issue.
I use the following pipe:

  ## Tx
gst-launch-1.0 v4l2src device=/dev/video2 !  video/x-h264,width=1280,height=720,framerate=15/1 , profile=main  ! h264parse config-interval=15 ! filesink location=/dev/stdout | sudo nice -n -9 sudo /home/pi/DroneBridge/video/video_air -d 8 -r 4 -f 1024 -t 2 -b 18 -c 200 -a 0 -n 00cxxxxxxx

  ## Rx
sudo /home/pi/DroneBridge/video/video_gnd -d 8 -r 4 -f 1024 -c 200 -p N -v 5000 -o -n 00cxxxxxxx | /home/pi/DroneBridge/video/pi_video_player/db_pi_player_48

would be nice if anybody has an idea
thx

seeul8er, 2022-01-09:
The video bitrate might be too high. Try lowering the video bitrate it or try increasing the DroneBridge transmission bitrate (try -b 54 e.g. - In that case you will also need to increase the bitrate when setting the cards into monitor mode. The -b flag is often ignored by drivers.

hex0r, 2022-01-28:
your are right, there some issue with the bitrate unfortunately i could not solve it at the moment.
Can i change the bitrate with an command without changing it in the config and restart the service?

seeul8er, 2022-01-28:
You will have to kill/stop the video_air part with the pipe from the video source. Then run it with an updated parameter set

hex0r, 2022-01-29:
yes i do it that way, but you mentioned 
"you will also need to increase the bitrate when setting the cards into monitor mode. The -b flag is often ignored "
how can I change this?
Is this set by the startup python script ?