DroneBridge/DroneBridge #86: v0.6 .img improvements (shrink)

AlexShmelev, 2022-08-27:
Hi, i was wondering, why does image size is 16 GB ?
I looked partable of .img file and file size on partitions and I see 
```
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      277MB   2689MB  2412MB  primary  ext4
 3      2689MB  15,9GB  13,2GB  primary
```
The third partition is ExFAT (why?),  using 13GB and contain files only for 18MB!
```
/dev/loop29p1              253M   45M  208M  18% /mnt/loop1
/dev/loop29p2              2,2G  1,7G  401M  82% /mnt/loop2
/dev/loop29p3               13G   18M   13G   1% /mnt/loop3
```

Writing this image to SD card with `dd` will take "insane" amount of time for writing this empty space. 

Why don't just make this partition 100 MB, and resize by script on first boot to end of disk ? (Is this OS limitation where debian can't resize ExFat parts? Why not use FAT32 instead ?) 
Why does ExFAT needed ? Can't those file be stored on ext4 in `/DroneBridge` Folder instead of mounting this ExFAT partition ? 

Thanks.

seeul8er, 2022-08-28:
None of the images you can download has 16GBs in size. Not sure what you mean. On the first boot, the image will expand the third partition to the maximum available size just as you suggested.
As far as I remember the third partition is for storing configuration data and the recorded video (not sure that is already implemented in v0.6). That is why it is that big and exFAT. exFAT can be read by all OS and the video recorder does not need to split the files in 4GB junks, as it would be the case with the outdated FAT format.

16GB are a pain in the ass when copying the entire SD card, but other than that it does not have a disadvantage.

AlexShmelev, 2022-08-29:
>None of the images you can download has 16GBs in size. Not sure what you mean. 

https://github.com/DroneBridge/DroneBridge/releases/tag/v0.6.0
[Get the ready to use image for the Raspberry Pi here](https://forstudents-my.sharepoint.com/:u:/g/personal/ga25puh_forstudents_onmicrosoft_com/EZah1nq0VhNKtHNdDhQNh5QBtVIsj1dOsqOnqJYzGeuN8w?e=S7vCbh)

Download link on 5GB archive with 16GB .img file in it.