DroneBridge/DroneBridge #92: Trying to use DroneBridge on AML-S905X-CC (Le Potato)

ccardiff, 2023-04-19:
Due to inability to purchase RPi 3 or 4 here in Canada I bought AML-S905X-CC (Le Potato) boards to use with my FPV. 

I have followed the instructions on wiki but get this error. 

gndpi@raspberrypi:~/DroneBridge $ sudo ./start_db start
./start_db: line 45: /home/pi/DroneBridge/syslog_server/db_syslog_server: No such file or directory
./start_db: line 47: vcgencmd: command not found
./start_db: line 58: /tmp/undervolt: Permission denied
./start_db: line 61: /usr/bin/vcgencmd: No such file or directory
Building DroneBridge modules...
./start_db: line 63: cd: /home/pi/DroneBridge: No such file or directory
DroneBridge is already compiled for the CPU architecture
head: cannot open '/home/pi/DroneBridge/db_version.txt' for reading: No such file or directory
Welcome to DroneBridge v Beta (GND) -
python3: can't open file '/home/pi/DroneBridge/startup/init_wifi.py': [Errno 2] No such file or directory
python3: can't open file '/home/pi/DroneBridge/startup/start_db_modules.py': [Errno 2] No such file or directory

dos2unix: /DroneBridge/osdconfig.txt: No such file or directory
dos2unix: Skipping /DroneBridge/osdconfig.txt, not a regular file.

./start_db: line 27: cd: /home/pi/DroneBridge/osd: No such file or directory
Building OSD:
[  4%] Built target gf256
[ 27%] Built target db_common
[ 37%] Built target sumd_test
[ 39%] Built target fec_speed_test
[ 50%] Built target control_ground
[ 56%] Built target control_air
[ 60%] Built target db_status
[ 64%] Built target db_proxy
[ 72%] Built target video_air
[ 81%] Built target video_gnd
[ 85%] Built target db_recorder
[ 91%] Built target usbbridge
[ 95%] Built target db_syslog_server
[100%] Built target readrc
./start_db: line 33: ./osd: No such file or directory

Is the situation hopeless or can somebody point me in the right direction?

Thanks,

seeul8er, 2023-04-19:
The ready made images with many of the camera and OSD related things are only compatible with the raspberry pi. It is possible to run it on other platforms (except for the OSD), however you will have to do the debugging yourself. You can not use the ready-made images and expect them to just work.

Some hints:
`/home/pi/DroneBridge/syslog_server/db_syslog_server`
Seems like that file is missing or the path is not existing. Try coping it to that location.

`vcgencmd: command not found`
Raspberry Pi specific command.

`/tmp/undervolt: Permission denied`
Raspberry Pi specific bit

`/home/pi/DroneBridge`
folder seems not existing. You need to create user "DroneBridge" and copy the files there.

As you see there is a lot of info to get things fixed. You will need to do that without my support, since there are simply too many SBCs out there and I can not support them all.

kda186, 2024-01-22:
I completed dronebridge work on RK3588 with reference

https://dronebridge.gitbook.io/docs/developer-guide/downloading-and-compiling

It should be noted that if you use OSD, you may not be able to complete the compilation, because OSD uses the strong association module of Raspberry Pi, and compilation cannot be completed directly in other architectures.