DroneBridge/DroneBridge #95: ESP32-S3-WROOM1: Access Point Not Showing After Flashing DroneBridge Firmware

stevejohn163, 2024-09-05:
I flashed the DroneBridge firmware onto my ESP32-S3-WROOM1, and while the upload was successful, the SSID for the access point does not appear for me to connect and configure DroneBridge settings.

Steps taken:
1. Erased the flash and re-uploaded the firmware, but the issue persists.
2. Tried using another ESP32-S3-WROOM1 board, and the same issue occurred.

The access point is not discoverable, so I cannot apply any settings in either client or AP mode. Any ideas on how to resolve this?

seeul8er, 2024-09-06:
Wrong project ->https://github.com/DroneBridge/ESP32
I have the same board and the latest release flashed via the online flasher works fine for me. [Check the logs](https://dronebridge.gitbook.io/docs/dronebridge-for-esp32/logging-and-debugging)

closed 2024-09-06 by seeul8er

stevejohn163, 2024-09-06:
I also tried uploading the firmware using the online flasher, but the SSID still doesn't show up. I'm not getting any output on the serial monitor either—just a solid red LED on the board. I'm not sure what I'm doing wrong. As mentioned earlier, I completely erased the board and re-uploaded the firmware, but there's still no output on the serial monitor, no SSID, and just the solid red LED on my ESP32-S3 board.